rust-embedded/heapless #2: Are there any other crates like this one?

XOSplicer, 2017-08-11:
I forked the create as I think it would be a good exercise for me to work a bit with rust.
Just a quick question: Do you know about any crates that provide a similar approach?
I know of https://github.com/dropbox/rust-alloc-no-stdlib which could be used as the underlying allocator for this lib.

japaric, 2017-08-13:
There are a few crates that provide `Vec` like data structures that need no memory allocator like [arrayvec](https://crates.io/crates/arrayvec) but I think you can only allocate the growable collection on the stack. OTOH, `heapless` lets you allocate growable collections on static memory.

> I know of https://github.com/dropbox/rust-alloc-no-stdlib which could be used as the underlying allocator for this lib.

This crate doesn't use a memory allocator.

XOSplicer, 2017-08-13:
> This crate doesn't use a memory allocator.

Sure, sorry for the missunderstanding. You of course have to supply a (statically) allocated array to the ::new() constructors.

You can check out my progress on this crate if you have the time [here](https://github.com/XOSplicer/heapless)

japaric, 2017-08-22:
BTW, I'm pretty sure that CircularBuffer.deref will become unsound once you add some method to remove elements (for example: create empty CB, push 2N (N == capacity) elements, pop the last, call deref) so it may be best to remove it.

XOSplicer, 2017-08-23:
Yes I also though of that. Using its content with an iterator only would be the best IMO

closed 2017-09-28 by XOSplicer