rust-embedded/heapless #6: Improve / refactor `RingBuffer` iterators

japaric, 2017-10-31:
The implementation of `ring_buffer::{Iter,IterMut}` is pretty much a copy of the other. The duplication can be eliminated by using a macro (see implementation of iterators in libcore). Also both `Iter` and `IterMut` use the default methods provided by the `Iterator` trait; the `Iterator` implementation for those two could override the default methods for better performance.

japaric, 2018-04-08:
Triage: Still needs to be implemented.

XOSplicer, 2018-04-09:
Hi @japaric, 
check out https://github.com/XOSplicer/heapless/commit/57452a8138f70eeeb8bffcef51b1a870286cf06c if you have the time.
I have never worked with macros before, so some things might be a little rough.
If you like the changes, I can open a PR.

commit a8d404fb (2018-04-19): Merge #26

26: Ringbuffer iterators macro r=japaric a=XOSplicer

Resolves #6 

Refactor the ring_buffer::{Iter,IterMut} implementations to use a macro, so there is less duplicate code.
Inspired by the[ slice iterator macro in lib core](https://github.com/rust-lang/rust/blob/master/src/libcore/slice/mod.rs).

I have never worked with macros to generate const / mut implementations.
So this might be a little rough.

`#![allow(stable_features)]` is needed for current rust nightly, since otherwise `conservative_impl_trait` would throw a warning. Removing the feature flag would not be backwards comp [... cut]
PR #26 (2018-04-19): Ringbuffer iterators macro
Resolves #6 

Refactor the ring_buffer::{Iter,IterMut} implementations to use a macro, so there is less duplicate code.
Inspired by the[ slice iterator macro in lib core](https://github.com/rust-lang/rust/blob/master/src/libcore/slice/mod.rs).

I have never worked with macros to generate const / mut implementations.
So this might be a little rough.

`#![allow(stable_features)]` is needed for current rust nightly, since otherwise `conservative_impl_trait` would throw a warning. Removing the feature flag would not be backwards compatible with earlier nightly versions.
closed 2018-04-19 by bors[bot]