rust-embedded/heapless #56: impl Debug

TeXitoi, 2018-09-01:
Having the structures implementing Debug will help debugging.

XOSplicer, 2018-09-04:
I agree with Debug, Default, Eq, PartialEq and Hash. However most of them are available through `Deref<Target=[T]>` on Vec<T>. Clone on the other hand would not work as the typical heap allocated clone but would yield a new stack allocated instance. If @japaric doesn't have the time, I can look into this

TeXitoi, 2018-09-04:
The big blocker is that I can't derive when my strict contain a heapless object.

For clone, the semantic is valid, and don't forget that any Copy is Clone. It would be possible to derive Copy on these objects, bit I don't think it would be great semantically.

XOSplicer, 2018-09-05:
Sure go ahead, you might want to check https://rust-lang-nursery.github.io/api-guidelines/interoperability.html for a list of Traits that should / could be implemented for every of the public data types

XOSplicer, 2018-09-10:
I tried to implement some of the traits:
https://github.com/XOSplicer/heapless/pull/8/files
however I were hitting some things to decide upon: 
- BinaryHeap has no `Eq` implementation in the standard lib, shoult it have one?
- Implementing `Ord` for Vec is not straight forward
What are your thoughts on that?

TeXitoi, 2018-09-10:
`Eq` is not implementable for `BinaryHeap` as the same elementx can be in different order depending on the insertion order.

For `Ord` on `Vec`, you can just forward to `impl<T> Ord for [T] where T: Ord`.

XOSplicer, 2018-09-10:
i guess when `Eq`, is not implemented for `BinaryHeap`, so is `Hash` aswell not.

Well the problem I was facing with `Ord` for `Vec` is that one should be able to compare a `Vec<T, N>` against a `Vec<T,M>` where N and M are static capacities. However I failed at expressing that successfully in the trait implementation. Maybe you come up with an implementation for that one. A similar implementation for `String<N>` would be the next step, as the same problem is present there.

TeXitoi, 2018-09-10:
For `Ord` between `Vec<T, N>` and `Vec<T, M>`, the `Ord` trait doesn't allow it as lhs and rhs must be the same type.

sanmai-NL, 2018-09-21:
@TeXitoi: you wrote it would not be great to derive `Copy`. Could you explain a bit? I understand the cost of a copy may be larger than e.g. for the common `Copy` types. But people using this library would know that (certainly if this is explained in the docs)? If `heapless::String` were `Copy`, it would be very useful for error handling/backtraces.

TeXitoi, 2018-09-21:
My point is that, as `std::Vec` and `std::String` are not copy, heapless version should not, else there will be different behavior (i.e. copy instead of move).

Why would it be very useful for error handling and backtraces?

sanmai-NL, 2018-09-21:
`heapless::Vec` and `heapless::String` are intended to be API compatible with their `std` counterparts, the goal is not for them to have the same memory allocation behavior (indeed the point is for them _not_ to).

It would be useful to be able to render a backtrace in the shape of a `Copy` field of type `heapless::String` within some `E` type, finally within a `core::result::Result`. Right now, `std::io::Error` isn’t even `Clone` so error values of this type can’t be propagated from out of e.g. `Fn` closures. Any `std::error::Error` types that contain such a value somehow, e.g. as part of a backtrace, can’t be `Clone` either. Also, since many common `std::error::Error` types aren’t `Copy`, moving them is costlier, so if one’s `std::error::Error` type is propagated up a call stack as part of a backtrace, this has a higher overhead because of the heap memory involved.

TeXitoi, 2018-09-22:
I really don't get your point. A move is always a memcpy.

Even of heapless objects are not copy, these is no memory allocation when moving them around.

sanmai-NL, 2018-09-22:
There’s two things I claimed and it seems you do not discern them in your interpretation.

1. It’s helpful for `heapless::String` to be ~`Copy`~ at least `Clone`
If so, `heapless::String`s can be used in e.g. an `Error` type that can, e.g., be moved into an `Fn` closure via a `Result`. It is true that the `Copy` vs. `Clone` distinction is only relevant to express to cost of moving, copying or cloning such an object. AFAIK, there is potentially a stack memory allocation involved when _returning_ a heapless object.

2.  (A) Using a `heapless::String` in an `Error` type instead of (B) an `Error` type that carries `alloc::string::String` fields would be more efficient e.g., for the purpose of generating some sort of backtrace,

If irrespective of whether `heapless::String` gains `Copy`, `Clone` or neither, we compare approaches A and B, then `alloc::string::String` is stored in both the stack and the heap, whereas a `heapless::String` is only stored on the stack. Moving, copying or cloning them can have a different cost for A vs. B.

Please let me know if we are on the same page or not. I’m happy to learn from it.

sanmai-NL, 2018-09-22:
@TeXitoi: I fully agree with you, and I also have to clear up some confusion I had. 

Indeed, it would not be wise to pass `String`s as fields of an `Error` type, not even the heapless implementation. The amount of bytes to copy would be large in some cases (move, `Copy` and certainly `Clone`). I understand from https://doc.rust-lang.org/book/second-edition/ch04-01-what-is-ownership.html#ways-variables-and-data-interact-move that a move of an `alloc::string::String` does not copy the heap memory (of course), only a `Clone` would. A heapless `String`, however has to be copied from the stack upon a move. So in a sense, returning (i.e. moving) a `heapless::String` may be costlier than returning an `alloc::string::String`. 

I floated an idea where I hoped I could cheaply append cause upon cause to a `heapless::string::String` buffer at each call site, to construct a backtrace. It wasn’t thought through well enough, since a `heapless::string::String` is not (necessarily?) pointer to some stable memory but it is contained in stack frame (except if `'static`). It has to be copied when it is being returned to the caller onto the caller’s stack frame. And as you already suggested, size that has to be copied is larger than or equal to `alloc::string::String`’s.

And last but not least, all of this is irrelevant in practice, because `Error` types are `Display` and there is no need to carry a textual description as a field even in the happy path, not even if you would keep the ‘cause’ error as field (nested, for backtracing).

One question after all this thought. Is it possible to create an owned, statically allocated, heapless, immutable `String`?

TeXitoi, 2018-09-22:
For your last question, a `&'static str` is what you want, no?

sanmai-NL, 2018-09-22:
No, I am wondering whether there’s a way to have an _owned_ object that you can `Copy` into an `Fn` closure. No moving, no cloning, but an owned value backed by a `&'static str`. `std::borrow::Cow` would not give me an owned object without copying or cloning, right? I am curious about e.g. a `const fn` that can initialize an owned `String` from a `str` literal, a `String` which implements `Copy` but won’t be copied in reality.

MabezDev, 2019-01-30:
Is there any movement on this? I'm particularly keen to get fmt::debug for Queue.

XOSplicer, 2019-01-31:
> Is there any movement on this? I'm particularly keen to get fmt::debug for Queue.

- I there is a PR open for the `Clone` implementation #68
- `Debug` seems to be implemented for all data structure except for `LinearMap`. 
- The comparators are still up for discussion.
- `Hash` does not seem to be implemented for any of the data structures

XOSplicer, 2019-02-10:
This is the current (master, unreleased after v0.4.1) implementation situation:

|               | Default | Debug | Clone | Hash | comparators |
|---------------|---------|-------|-------|------|-------------|
Vec             |         | x     | x     | x    | Eq, PartialEq |
String          |         | x     | x     | x    | Eq, PartialEq |
LinearMap       |         | x     | x     |      | Eq, PartialEq |
IndexMap        | x       | x     | x     |      |             |
IndexSet        | x       | x     | x     |      | PartialEq   |
BinaryHeap      |         | x     | x     |      |             |
Queue           |         | x     |       |      |             |

We can certainly move ahead and implement default for all the data structures. The other traits should be implemented where applicable.

XOSplicer, 2019-02-11:
Some thoughts on this:

- LinearMap, IndexMap, IndexSet and BinaryHeap do not have a defined iteration order.  Therefore implementing `Hash` is not straight forward.

- Queue has type parameters for the `Atomic` and `XCore` to use, so I don't know if there should be a default implementation for it. A default implementation per type variant might be feasible.

- Implmenting `Eq` for the map types might be possible (see https://doc.rust-lang.org/std/collections/struct.HashMap.html#impl-PartialEq%3CHashMap%3CK%2C%20V%2C%20S%3E%3E). 

- For the Heap in std `Eq` is not implemented, so  don't think it should be in this crate

japaric, 2019-12-17:
I believe this has been largely done. Any remaining items should be discussed in separated issues.

closed 2019-12-17 by japaric