rust-embedded/heapless #65: Interest in adding BBQueue for batched SPSC support

jamesmunns, 2019-01-06:
Hey @japaric, I've gotten [`bbqueue`](https://github.com/jamesmunns/bbqueue) to a pretty usable state, and I was wondering if you would be interested in adding it to the `heapless` collections. It is a SPSC queue, however rather than being "one item at a time", it's more like "one slice at a time", meaning you can request a "chunk" of space to read/write, which pairs well with DMA space allocations.

Originally, I planned to use `generic-array` to match the interfaces provided by `heapless::queue`, however after struggling with getting that to work for my use cases, I ended up writing a "static allocator" [macro](https://github.com/jamesmunns/bbqueue/blob/master/core/src/lib.rs#L577-L592) based on your `singleton!()` macro in the `cortex-m` crate.

Let me know if you'd be interested in adding this, or if you have any thoughts or comments in the impl or what it would take to merge it in. I saw your other PR regarding DMA, I think this might work well with that (it gives you &'static [u8]s you can give to the DMA impl).

japaric, 2019-01-09:
Looks interesting! I'd like to review the code / API first. If I find any issue I'll report it to the bbqueue repo. Once I review the crate I'll comment here again.

jamesmunns, 2019-01-09:
Sounds good! I would actually really appreciate the review from you.

Right now there is one big soundness hole (https://github.com/jamesmunns/bbqueue/issues/14), which I plan to "fix" by making the `BBQueue::new()` method unsafe, until `Pin` has been stabilized, where I will probably change up the interface a bit to make sure the BBQueue has been Pinned before you can call `::split()`.

korken89, 2019-04-24:
Has this progressed anything lately?

jamesmunns, 2021-06-29:
I think these are likely to live on as separate crates. If there is interest in merging, feel free to ping me.

closed 2021-06-29 by jamesmunns