rust-embedded/heapless #102: heapless on const generics: status

burrbull, 2019-06-26:
I tried to port `heapless` types from GenericArray to `usize` const generics.
[Here](https://github.com/burrbull/heapless/tree/const-generics) is branch.

Today I managed to compile it in release mode.
For testing you can use: `cargo test --release`.

Issues:

* In debug mode compile fails.
* `Queue` doesn't work. Fails on type inference. I think the problem is in the interaction of default types and const generics. `pub struct Consumer<'a, T, U = usize, C = MultiCore, const N: usize>`
* I don't know how to set `PowerOfTwo` restriction for `indexmap`.

Everything else is pretty good.

XOSplicer, 2019-06-27:
Hi, 

this is a very interesting and strategical step forward for the heapless crate.
Porting heapless to const generics was on my mind aswell, but time is limited.

Here are my thought on the `PowerOfTwo` restriction:

In general what is needed is a static assertion for the const generic.

- There exists a thread which discusses exactly this feature as a language feature, but to no conclusion (Link)[https://internals.rust-lang.org/t/idea-static-assertions/7753/6]
- The `const_assert` macro of the (static_assert crate)[https://crates.io/crates/static_assertions] might help.
- An internal new type `Foo<const N: usize>(PhantomData)` with a trait `PowerOfTwoMarker` that is only implemented for `Foo<2>`, `Foo<4>`, `Foo<8>`, `Foo<16>` and so on might be possible, but limits the allowed values to an enumerable range for which this trait is imlememted. (I'm not sure about the PhantomData part though)

japaric, 2019-07-11:
@burrbull thanks for trying this out. I think the compiler team will appreciate all the bug reports.

From my side, I would prefer not to make heapless depend / use const generics in any form (even behind a Cargo feature) until the feature has been (partially) stabilized. Nightly features are a maintenance burden that I would prefer to avoid.

burrbull, 2019-10-22:
It's a good day.
https://github.com/rust-lang/rust/pull/65652
https://github.com/rust-lang/rust/pull/65092

korken89, 2020-09-23:
See #168 for up to date version of this.

closed 2020-09-23 by korken89