rust-embedded/heapless #109: peek_mut for BinaryHeap?

equal-l2, 2019-07-22:
`peek_mut` of std::collections::BinaryHeap is useful, for example we can use it to swap the top element of the heap efficiently, while `pop` and `push` may require more time because of `pop`.
I'd like to see that in heapless::BinaryHeap, too.

korken89, 2019-08-28:
Hi @japaric - this seems as a very useful feature.
Do you see any issues with this/what interface to use? I could churn out the implementation for it.
I'm thinking something like:
```rust
myheap.update_top(|current_top| /* do changes to current top */ );
```

korken89, 2019-08-28:
I did an implementation in https://github.com/japaric/heapless/pull/117

japaric, 2019-08-28:
if the API is in `std::BinaryHeap` then you can just send a PR with the same API. This applies to all collections ported from `std`.

> I did an implementation in #117 

The `std` API also resorts the heap (in `PeekMut`'s destructor). Let's use that API instead of designing a new one.

korken89, 2019-08-28:
Makes sense, I'll change to that

korken89, 2019-08-28:
@japaric it's fixed now

korken89, 2019-08-29:
Hi, this is now added!
However, I am unable to get the pop API to work and need to look into it:

```
Error[E0223]: ambiguous associated type
   --> src/neighbour/mod.rs:163:17
    |
163 |                 BinaryHeap::PeekMut::pop(value);
    |                 ^^^^^^^^^^^^^^^^^^^^^^^^ help: use fully-qualified syntax: `<heapless::binary_heap::BinaryHeap<_, _, _> as Trait>::PeekMut`
```

jonas-schievink, 2019-08-30:
Don't you want something like `binary_heap::PeekMut::pop` instead?

korken89, 2019-08-30:
Haha, well this is awkward - that is exactly what needed to be done :)
Thanks for the hint!

korken89, 2019-08-30:
I'll also close this as the requested feature has been added

closed 2019-08-30 by korken89