rust-embedded/heapless #111: Explain recommended strategy for moving data interrupt <-> main

birkenfeld, 2019-08-06:
It would be nice to have a little more fleshed out info about the recommended ways to share data between interrupt handlers and the main program.  With the addition of Qxx, I assume that this is to be preferred, but its size limitation makes it not always applicable if items come in bursts and handling them can take a while.

Using Queue in such a context still requires `static mut` unsafe, and also it isn't explained if `split()` can be called in the main program in a way that it doesn't require `interrupt::free` to en/dequeue things.

Sorry, this is a bit rambly, but these are some of the things I was expecting to find in the docs, since this crate seems like it's tailored for these tasks.

korken89, 2020-09-23:
This is usually up to the framework that is being used for resource handling.
E.g. RTIC has examples for how this should be done, while in the raw `cortex-m` world it's a bit more ambiguous with multiple different solutions.

closed 2021-05-23 by korken89

birkenfeld, 2021-05-23:
And it's not possible to mention those solutions? RTIC is very nice but feels always teetering in the brink of being abandoned, and I would wager that most people don't use it.

korken89, 2021-05-23:
@birkenfeld here is an example: https://github.com/rtic-rs/cortex-m-rtic/blob/master/examples/late.rs

Not sure what you mean with that RTIC is being abandoned? It's actively maintained and developed.

birkenfeld, 2021-06-27:
You're right - I guess pace has picked up again.