rust-embedded/heapless #112: "Classic" ringbuffer support?

birkenfeld, 2019-08-06:
I've needed a "classic" ringbuffer quite a few times now, and since it's not that hard rolled my own.  What I mean is:

* fixed capacity array plus index of "head"
* push() moves head and overwrites
* allows random access using iterator, and some standard things like first(), last()
* (maybe:) all items are initially filled with some default, allows getting rid of Option<>s for first() etc
* normal `&mut` methods, no lock-free support

This is nothing fancy, but seems like it falls under the mission statement of `heapless`?  I'll contribute an initial implementation PR, if yes.

korken89, 2019-08-14:
Hi, the `Queue` seems to do what you want through the `Producer` and `Consumer` interface.
Or is there an issue there that you are facing?

birkenfeld, 2019-08-14:
On a glance it seems similar to a Queue, but I need access to all elements without popping them.

Think a device with a noisy sensor that reads the sensor every millisecond and keeps the last N readings to make an average, minimum etc. calculations.

So the main differences to Queue are:

* access to all elements at any time, in exchange, no lock free characteristic required
* if full, oldest entries are just overwritten without needing to be popped
* (optional) always filled to capacity with a "null" value, see the last point in OP

korken89, 2019-08-14:
Ah, I see. Lets see what @japaric says :)

japaric, 2019-12-17:
Sorry for missing this issue.

This sounds reasonable to me. The `Default` bound to avoid the `Option` makes sense as well. 

> plus index of "head"

Is this is a push-only ring buffer?

> allows random access using iterator, and some standard things like first(), last()

Maybe also an `as_slice` method? To get access to all elements in arbitrary order. `rb.as_slice().iter().cloned().sum()` may eval faster than `rb.iter().cloned().sum()`.

birkenfeld, 2019-12-18:
> Is this is a push-only ring buffer?

Yes. To avoid confusion with real ring buffers I've renamed it to `HistoryBuffer` in the PR. (Another name would be something with `Window`...)

> Maybe also an as_slice method?

I've implemented `AsSlice` and `AsMutSlice`.

commit 4d613da7 (2020-04-30): Merge #136

136: add the HistoryBuffer type r=korken89 a=birkenfeld

Closes #112

Unfortunately `const` construction won't work since the `Default` or `Clone` bound on `T` can't be used then, so I've not added an internal type to `i.rs`.

Let me know if you think any other traits should be implemented...
PR #136 (2020-04-30): add the HistoryBuffer type
Closes #112

Unfortunately `const` construction won't work since the `Default` or `Clone` bound on `T` can't be used then, so I've not added an internal type to `i.rs`.

Let me know if you think any other traits should be implemented...
closed 2020-04-30 by bors[bot]