rust-embedded/heapless #114: Comment in the pool! macro does not propagate to the struct

korken89, 2019-08-14:
Example:
```
pool!(
    /// Pool allocator for the DMA engine running the transfers, 
    /// where each node in the allocator is an `MyPoolNode`
    #[allow(non_upper_case_globals)]
    MyPool: MyPoolNode
);
```

The attribute propagates properly, but the comment does not.
This makes it difficult to document and show examples connected to the pool.

burrbull, 2019-08-14:
how about `doc` attribute?

korken89, 2019-08-14:
It falls down on the internal static, and is not visible to the public struct.
I gave it a try to fix like this, but the macros complain about ambiguity:

```
#[macro_export]
macro_rules! pool {
     ($(#[doc = $doc:expr])* $(#[$($attr:tt)*])* $ident:ident: $ty:ty) => {
        $(#[doc = $doc])*
        pub struct $ident;

        impl $crate::pool::singleton::Pool for $ident {
            type Data = $ty;

            fn ptr() -> &'static $crate::pool::Pool<$ty> {
                $(#[$($attr)*])*
                static $ident: $crate::pool::Pool<$ty> = $crate::pool::Pool::new();

                &$ident
            }
        }
    };
}
```

Not sure how to get around that.

closed 2021-05-23 by korken89