rust-embedded/heapless #116: Serde support not working?

vagola, 2019-08-23:
I am trying to derive structure and getting the following error:

```
the trait bound `heapless::string::String<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UInt<typenum::uint::UTerm, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B1>, typenum::bit::B0>, typenum::bit::B1>>: config::_IMPL_DESERIALIZE_FOR_Config::_serde::Deserialize<'_>` is not satisfied
```

It's strange because i see that heapless supports serde

birkenfeld, 2019-08-23:
Serde support is optional; did you activate the `serde` feature for `heapless`?

vagola, 2019-08-23:
In cargo.toml there are no features. How to enable it? Something like that `heapless = { version = "0.5", default-features = false, features = ["serde"] }`?

birkenfeld, 2019-08-23:
Yes, that should work.

In heapless' `Cargo.toml` there is no explicit feature since the dependency is marked `optional = true`, which creates such a feature automatically.

vagola, 2019-08-23:
Thanks for the help!

closed 2019-08-23 by vagola