rust-embedded/heapless #123: heapless does not build for riscv32imc-unknown-none-elf target

rjoleary, 2019-11-15:
rv32imc does not have atomics leading to this issue:

```
error[E0599]: no method named `compare_exchange_weak` found for type `core::sync::atomic::AtomicPtr<pool::Node<T>>` in the current scope
   --> /home/ryan/.cargo/registry/src/github.com-1ecc6299db9ec823/heapless-0.4.4/src/pool/mod.rs:278:33
    |
278 |                 match self.head.compare_exchange_weak(
    |                                 ^^^^^^^^^^^^^^^^^^^^^ method not found in `core::sync::atomic::AtomicPtr<pool::Node<T>>`
```

jamesmunns, 2019-12-17:
I believe this would be closed by #120.

japaric, 2019-12-17:
> I believe this would be closed by #120.

That PR is not sufficient. That target does not have atomic *loads* either so `AtomicUsize` et al. are missing from `core` meaning that the `spsc` module doesn't compile for this target even after disabling the "cas" Cargo feature.

commit 940d2e9d (2019-12-17): remove spsc / pool / mpmc modules on targets w/o atomic / CAS support

closes #123
commit e4abb7e1 (2019-12-17): Merge #135

135: remove spsc / pool / mpmc modules on targets w/o atomic / CAS support r=japaric a=japaric

closes #123
commit 44c66a74 (2019-12-18): remove spsc / pool / mpmc modules on targets w/o atomic / CAS support

closes #123
commit 3fdd3e86 (2019-12-18): Merge #135

135: remove spsc / pool / mpmc modules on targets w/o atomic / CAS support r=japaric a=japaric

closes #123
closed 2019-12-18 by bors[bot]