rust-embedded/heapless #139: Allow changing the `len` type of `Vec`

jonas-schievink, 2020-01-09:
`Queue` supports switching out the type to use for head and tail pointers, which can save memory (instead of 8 Bytes for 2 `usize`s, I can choose to pay only 2 Bytes for 2 `u8`s). It would be nice to have the same option for `Vec`, which currently always uses `usize`.

YuhanLiin, 2021-04-06:
I looked into implementing this using a generic length type and I'm running into issues initializing the generic length to 0. The constructor for `Vec` is a `const fn` so I can't initialize the length with any trait methods. I might look into using macros like `Queue` does to implement most methods.

korken89, 2021-04-29:
With the `const generics` MVP and found soundness issues, I have decided to keep `heapless` simple and easier to show correctness.
Hence I will not be pursuing these micro optimizations for the time being.

closed 2021-04-29 by korken89