rust-embedded/heapless #141: Question mark semantics for push on heapless vector.

treyfortmuller, 2020-01-16:
I work in a crate that requires static allocation everywhere, `heapless` has been great. I do have a lot of code that looks like:

```rust
let mut heapless_vec: Vec<i32, U32> = Vec::new();

heapless_vec.push(3).expect("Failed to push");
```

Now I'd like to get all these pushes on the same level of error forwarding semantics as the rest of my crate and avoid any possibility of a panic. Because `push<T>` returns a `Result<(), T>` rather than a `Result<(), &'static str>`, we can't use the brilliant `?` operator to pass errors from pushing up the error chain.

Right now in order to return the error I want, I have to do something along the lines of

```rust
let mut heapless_vec: Vec<i32, U32> = Vec::new();

match heapless_vec.push(3) {
    Ok(_) => (),
    Err(_) => return Err("failed to push onto heapless vec."),
}
```

This is really verbose. If we had another type of `push` function that returned a `Result<(), &'static str>` then we could use `?` like this:

```rust
let mut heapless_vec: Vec<i32, U32> = Vec::new();

heapless_vec.push_msg(3)?;
```

This would represent a huge clean up and brings `heapless` more in line with encouraged Rust error handling semantics.

What are your thoughts on this?

treyfortmuller, 2020-01-16:
For reference I've written a macro which corrects the result unwrapping semantics, this is similar to the ``try!` macro which deprecated in favor of the `?` syntax:

```rust
macro_rules! try_push {
    ($expr:expr) => {
        match $expr {
            Ok(_) => (()),
            Err(_) => {
                let failed_msg: &'static str = "Failed to push an item to a heapless vector.";
                return Err(failed_msg);
            }
        }
    };
}
```

birkenfeld, 2020-01-16:
You can just do `vec.push(x).map_err(|_| "Failed to push")`.

XOSplicer, 2020-01-16:
The reason for having for having `Vec::<T>::push` return `Result<(), T>` is primarily to be able to retrieve the element that failed to insert, as it was passed by move semantics. Otherwise it would be dropped. 

There have been considerations to alter the return type to something in the lines of `Result<(), CapacityError<T>>`, where `CapacityError<T>` holds the item, but can also be used as a proper error, or at least be converted implicitly to an error. 

This approach however has never been followed through, as the `map_err` approach is convenient enough.

treyfortmuller, 2020-01-23:
Totally valid. Thank you for the tip on `map_err` I wasn't aware of that one!

closed 2020-01-23 by treyfortmuller