rust-embedded/heapless #146: Heapless paper pdf?

aep, 2020-04-05:
Is it possible to get access to the associated paper? "Heapless: Dynamic Data Structures without Dynamic Heap Allocator for Rust" somewhere?

japaric, 2020-05-03:
Hello @aep. Sorry for the late reply.

I believe the paper can only be accessed if you have access to IEEE publications (?). @perlindgren would know better.

That paper was written before heapless v0.5.0 was released so the implementation covered there (v0.4.x) looks rather different than the current implementation (iirc, the version in the paper used the `CoerceUnsized` traits and required nightly).

Is there any particular question you had about that paper?

aep, 2020-05-03:
well [zetz](https://github.com/zetzit/zz) is entirely heapless, so i'm interested in everything other people have written on the topic. In this case specifically i'm interested in anything you have learned that isn't obvious. Implementing stack based structures is pretty straight forward in zetz, but there are probably algos that fit better and some worse to being on the stack, when it comes to things like cache locality.

perlindgren, 2020-05-03:
You can contact me personally, and I can share a copy of the paper (its behind a paywall).

closed 2020-09-23 by korken89