rust-embedded/heapless #158: Cannot create static HistoryBuffer

andresv, 2020-05-05:
With `Vec` it is possible to do this:

```
use heapless::Vec; // fixed capacity `std::Vec`
use heapless::consts::U8; // type level integer used to specify capacity
static mut XS: Vec<u8, U8> = Vec(heapless::i::Vec::new());
let xs = unsafe { &mut XS };
```

However [HistoryBuffer](https://github.com/japaric/heapless/blob/master/src/histbuf.rs#L37) is missing in `heapless::i`.

birkenfeld, 2020-05-05:
That's (currently) unavoidable since the `T: Default` bound can't be used there.

andresv, 2020-05-05:
So currently I can create it only to stack?

korken89, 2020-05-05:
I will have a look at fixing this, it's was a miss in my review of the `HistoryBuffer` PR

korken89, 2020-05-05:
For now, you can put it in an option as a workaround.

andresv, 2020-05-05:
Like this?

```Rust
#[app(device = hal::stm32, peripherals = true)]
const APP: () = {
    struct Resources {
        ntc: &'static HistoryBuffer<u16, U100>,
    }

    #[init]
    fn init(mut ctx: init::Context) -> init::LateResources {
        // hack to make a static alloc
        static mut NTC: Option<HistoryBuffer::<u16, U100>> = None;
        unsafe {NTC = Some(HistoryBuffer::new())};

        init::LateResources {
            ntc: unsafe {NTC.as_mut().unwrap()},
        }
    }
```

korken89, 2020-05-05:
Ah no, I think this should work with RTFM:

```rust
#[app(device = hal::stm32, peripherals = true)]
const APP: () = {
    struct Resources {
        ntc: HistoryBuffer<u16, U100>,
    }

    #[init]
    fn init(mut ctx: init::Context) -> init::LateResources {
        init::LateResources {
            ntc: HistoryBuffer::new(),
        }
    }
```

andresv, 2020-05-05:
Indeed, this is much better.

PR #198 (2021-04-22): Const generics port
Closes #158
Closes #168
Closes #207
Closes #115
closed 2021-04-22 by korken89