rust-embedded/heapless #165: Fallback behavior for std environments

sffc, 2020-05-23:
Right now, most of the data structures in the `heapless` crate simply return an error status if they have reached capacity.

When architecting a library that is intended to run in both `std` and `no_std` environments, it may be desirable to fall back to heap allocation in those environments where heap allocation is available.  For example, if you can determine that 99.9% of cases require a vector to be length 5 or shorter, then you set that as the default length, but for border cases, in an `std` environment, you can fall back to a heap-allocated data structure.

Are you aware of prior art for such a feature?  Would you think that this would be appropriate for the `heapless` crate under an `std`/`alloc` feature?

Context: https://github.com/unicode-org/icu4x/issues/77

@Manishearth

Manishearth, 2020-05-23:
it might make more sense to use https://docs.rs/tinyvec/0.3.3/tinyvec/ with ArrayVec/TinyVec being cfg'd

XOSplicer, 2020-05-23:
smallvec https://docs.rs/smallvec/1.4.0/smallvec/ solves exactly that use case of spilling allocations to the heap when the vector grows too large

Manishearth, 2020-05-23:
Smallvec has unconditional spill, not "spill only when you have alloc"

korken89, 2020-09-23:
I am unsure how to pursue this issue, heapless was specifically made for systems without allocation and this is clear from the interface.
What this sounds like is as if the interface should change based on the environment it is in, or that there should exist a parallel interface which aliases to the `std::Vec`?