rust-embedded/heapless #166: generic-array v0.14

tarcieri, 2020-05-24:
We just bumped all of the RustCrypto crates to `generic-array` v0.14, which adds some nice features:

https://github.com/RustCrypto/traits/pull/95

It'd be great if `heapless` got updated too so we could share a single version of `generic-array` (which hasn't happened so far as the RustCrypto crates were previously stuck on v0.12).

We use `heapless` in the `aead` crate to provide a `Vec`-like buffer type on `no_std`. Here's an example:

https://docs.rs/aes-gcm/0.5.0/aes_gcm/#in-place-usage-eliminates-alloc-requirement

tarcieri, 2020-06-03:
It looks like `as-slice` needs to get updated to make this happen: https://github.com/japaric/as-slice/issues/6

burrbull, 2020-06-04:
It requires minimum Rust version of 1.41.0

tarcieri, 2020-06-04:
Yes, it requires an MSRV bump. For what it's worth, we've bumped all of the RustCrypto crates to MSRV 1.41 for this reason.

nickray, 2020-06-06:
> We use `heapless` in the `aead` crate to provide a `Vec`-like buffer type on `no_std`. Here's an example:

You might be interested in https://github.com/japaric/heapless/pull/164, if that gets reviewed/accepted. I'm also planning a `CString` type, but this could be a "minor" (patch) update.

Regarding MSRV, maybe https://github.com/RustCrypto/traits/issues/43#issuecomment-566768846 is applicable here ("depend on range instead of specific version") and would cater to all user groups.

tarcieri, 2020-06-11:
FYI, MSRV for `generic-array` v0.14 was just reduced to 1.36.0: https://github.com/fizyk20/generic-array/pull/102

korken89, 2020-09-23:
While I am positive to updating generic array, I am unsure if this is a breaking change in heapless.
I think we had similar discussions before in `as-slice` where it became a bit of a storm when we updated the generic-array dependency.

Darneas, 2021-01-20:
I am facing the need to upgrading `generic_array` to 0.14.0, due to the following [issue](https://github.com/fizyk20/generic-array/issues/97#issuecomment-752868879).

Without it, I am forced to stay on rustc 1.47.0, as 1.48. included this [this](https://github.com/rust-lang/rust/pull/71274/) and now causes a panic. Not a deal-breaker, but having to stay on an old rustc is not ideal.

I don't know about how pressing this is for others, but maybe heapless should bump it now to stay in sync with the rest of the ecosystem (rustc + generic_array)?

korken89, 2021-01-20:
It's a bit of a tough issue, this is a breaking change. I'm not sure how pliable `heapless` is to supporting both `0.13` and `0.14` of `generic_array`.
I was hoping we can hold out until `const generics` are released and make a major bump then and remove `generic_array` all together.
However if `generic_array` has UB in 0.13, I don't really see any option than to bump it.

korken89, 2021-01-20:
What do you think @japaric ? Should we bump to 0.14 within the current version of heapless to fix the UB?

Darneas, 2021-01-20:
Is [min_const_generics](https://github.com/rust-lang/rust/pull/79135) enough? 

If yes, it is already merged and rustc 1.51. is to land on March 25th 2021. So it is something one can plan for.

Though stripping `generic_array` would then also take some additional time.

korken89, 2021-01-20:
Yes, `min_const_generics` is enough. We have already ported `heapless` to it and are just awaiting stabilization

nickray, 2021-01-31:
@korken89 maybe it would be possible to bump heapless to 0.6 (bumping generic-array to 0.14), and then when you're ready, bump heapless to 0.7 (removing generic-array)? I know that's a few breaking change releases in a row, but it would be very helpful to not be stuck on 0.13 (+ the UB argument).

korken89, 2021-02-02:
I agree, we could do a release to fix the UB issue, but if it is UB I'd sort of want it to be in the current version.
Not sure if this makes sense, or if we should just go to a new version...

closed 2021-02-03 by korken89