rust-embedded/heapless #168: Const generics

kpp, 2020-07-11:
Const generics are getting more popular. I can try to implement `heapless` using const generics and push it into a separate branch until they are stable. It is a nice challenge both for coding skills and for const generics.

XOSplicer, 2020-07-13:
Reference  #102

burrbull, 2020-07-14:
@kpp Try if you can.
9 months have passed since my last attempt. Maybe something has changed in good way.

korken89, 2020-09-18:
@perlindgren has done the conversion, we are just awaiting stabilization to make a PR.

kpp, 2020-09-18:
How about make a WIP PR so no one else tries to do it on their own?

korken89, 2020-09-23:
@perlindgren Could you make a WIP PR with the conversion work you did?

korken89, 2020-09-23:
For reference the work is being done here: https://github.com/perlindgren/heapless/tree/const_generics

perlindgren, 2020-09-23:
Everything is implemented and the code passes all tests.

There are some type bounds that needs to be added for ensuring power 2, I don't recall exactly. These were provided by GenericArray previously, not sure if we want to define/implement them ourselves, or if these are available elsewhere.

tarcieri, 2021-02-03:
Perhaps I should ask on the `as-slice` repo, but there is a plan for it and const generics?

I ask because as of #189 `heapless` is now pulling in three versions of `generic-array` (v0.12, v0.13, v0.14)

korken89, 2021-02-03:
Technically, when const generics lands, all code related to `generic-array` will disappear as there is no longer any need for it.
This is also what we have done in the port of `heapless`. For `as-slice` I will do the same and remove all the code related to `generic-array` and specific array size implementations.

burrbull, 2021-02-03:
8 days before beta.

burrbull, 2021-03-26:
Time

korken89, 2021-03-26:
Here you can follow the work, we almost finished the port https://github.com/japaric/heapless/tree/const_generics

PR #198 (2021-04-22): Const generics port
Closes #158
Closes #168
Closes #207
Closes #115
closed 2021-04-22 by korken89