rust-embedded/heapless #171: Defmt support

theunkn0wn1, 2020-08-26:
I recently discovered the [`defmt`](https://github.com/knurling-rs/defmt)  library and attempted to integrate it into one of my embedded projects.

Unfortunately it appears that `heapless::Vec` doesn't implement the `defmt::Format` trait yet, which blocks `defmt` integration in my downstream. 

Feature request: support for `defmt::Format` in `heapless::Vec`

korken89, 2020-09-18:
Hi, I have commented on the PR itself.

korken89, 2021-05-23:
Available in 0.7.1

closed 2021-05-23 by korken89

mehmetalianil, 2021-07-01:
I get `error: unknown display hint: "[u8]"`  from:
```rust
let mut test_vec: Vec<u8, 128> = Vec::new();
   ... 
info!("the vector: {:[u8]}", test_vec);
```

Am I mistaken with the format specifier?

korken89, 2021-07-02:
Have you enabled the feature flag?

theunkn0wn1, 2021-07-11:
In my testing i used `{:?}` and `{=[?]}`, which both compile.

```rs
        let mut data: Vec<u32, 16> = Vec::new();
        data.push(0x42).expect("Failed to push into vec...");
        data.push(0x50).expect("Failed to push into vec...");
        data.push(0x10).expect("Failed to push into vec...");

        defmt::debug!("Writing data....");
        defmt::info!("{=[?]}", data);
        defmt::debug!("populating string...");
        let string: String<32> = String::from("I like pizza");
        defmt::debug!("Emitting string...");
        defmt::info!("{:?}", string);

```