rust-embedded/heapless #173: How to move spsc Producer and Consumer into different RTIC tasks?

cpfister, 2020-08-28:
I like the possibility to split an spsc queue into a Consumer and Producer value. However, when using RTIC, I run into lifetime problems trying to do that. My goal is to create a queue , split it, and move the producer to one RTIC task and the consumer to another RTIC task. So that the producing task only sees the enqueue function, while the consuming task only sees the dequeue function.

This means that I need to move the producer and consumer to separate RTIC resources (i.e., fields in the Resources struct of RTIC). Something like this:

#[rtic::app(device = hal, monotonic = rtic::cyccnt::CYCCNT)]
const APP: () = {
    struct Resources {
        prod: MyProducer<'static>,
        cons: MyConsumer<'static>,
    }

where MyProducer and MyConsumer are type-parameterized spsc Producer and Consumer, respectively.

These resources are then set up in the RTIC init task, something like this:

        let mut q: MyQueue = ... call spsc to create a new queue ...;
        let (p, c): (MyProducer, MyConsumer) = queue.split();
        init::LateResources {
            prod: p,
            cons: c,
        }

Unfortunately, here the queue q is dropped, but the prod and cons resources still live on, and internally contain references to q. So the borrow checker prohibits this move of p to proc, and c to cons:

error[E0597]: `q` does not live long enough
   --> src/bin/main.rs:118:120
    |
81  | #[rtic::app(device = hal, monotonic = rtic::cyccnt::CYCCNT)]
    |                                                            - `q` dropped here while still borrowed
...
118 |         let (p, c): MyProducer, MyConsumer) = q.split();
    |                                                                               ^^^^^^^^^^^^^--------
    |                                                                               |
    |                                                                                borrowed value does not live long enough
    |                                                                                argument requires that `q` is borrowed for `'static`

I guess that the queue should also be added to the resources, to make it live as long as prod and cons, but initialization syntax then does not allow to set up prod and cons. Something like this isn't syntactically valid:

        init::LateResources {
            queue: ... call spsc to create a new queue ...,
            (prod, cons): queue.split(),
        }

As I don't see a way around this problem, I'll probably give up on splitting the queue, but it's a pity that this separation of Producer and Consumer apparently only _almost_ works for me. Or is there a way, without using 'unsafe'?

nickray, 2020-08-28:
I'm not sure if avoiding unsafe is a hard requirement for you (why?), but the poor man's way of doing this involves a `static mut QUEUE: Option<Queue> = None` somewhere (e.g. in an initialization function), that you then initialize during setup with an actual queue that you can split, so you can add the producer and consumer to your late resources.

cpfister, 2020-08-30:
@nickray Hi Nicolas, thanks a lot for the quick reply, much appreciated!

I'll try your approach to better understand it hands-on, and for comparison purposes.

In the meantime, I have come up with a working version that does not split the queue. It works nicely, except that it would be better to expose just the producer-side of the queue to a producer task, and the consumer-side to a consumer task. This would make it immediately clear how the queue is used correctly.

Avoiding unsafe elements is not a hard requirement. The whole thing is an exercise for me to better understand the design space provided by Rust, regarding higher-level abstractions for embedded systems code. Providing safe abstractions on top of a hopefully thin layer of unsafe code is an eminently reasonable approach, e.g. allowing to concentrate code review efforts (and possibly more experienced developers) on the unsafe parts of the overall code. Having to give up such a desirable architecture quality so quickly would be somewhat disappointing.

cpfister, 2020-09-17:
@nickray I've now tried your recommendation with the static mut queue in an Option. I almost gave up, but after some experimentation it now works! Thanks for pointing out that approach!

The challenge was to split the static queue, and to move the resulting producer and consumer items into objects that are themselves RTIC resources, i.e., fields in the RTIC Resources struct. And so that they are not themselves wrapped in Options. To get working lifetime annotations was a matter of trial and error. I hope that at some point, I'll actually understand what I'm doing :-)

If there is interest, I can post the code that I came up with.

korken89, 2020-09-17:
Hi, sorry for missing this.
Here is the current recommended approach which does not require the use of an option.

```rust
#[app(device = some::device)]
const APP: () = {
    struct Resources {
        // Queue resources
        reader: Consumer<'static, MyType, U32>,
        writer: Producer<'static, MyType, U32>,
    }

    #[init]
    fn init(cx: init::Context) -> init::LateResources {
        static mut QUEUE: Queue<MyType, U32> = Queue(heapless::i::Queue::new());

        // Split queues
        let (writer, reader) = QUEUE.split();

        init::LateResources {
            reader,
            writer,
        }
    }

    // ...
}
```

cpfister, 2020-09-17:
@korken89 Thanks! I'll try to adapt my code to that approach - very nice if I could get rid of the option and the unsafe block in that way, and hide the queue in the init function!

cpfister, 2020-09-17:
@korken89 Almost worked. The remaining hurdle is that the compiler complains about my assignment to the QUEUE variable. It says "calls in statics are limited to constant functions, ...". The problem is that I don't directly use the heapless queues, but have wrapped them so that I can add additional state and processing (e.g. recording the maximum number of queue elements). So I have my own 'new' function.

Once the more powerful handling of const functions/expressions/generics in Rust becomes stable, this problem should vanish though, if I understand that effort correctly. (And let's dream of a heapless version that does not need the U32 etc. numtype hacks anymore, but where we can directly pass a normal constant as queue size argument when creating a queue...)

korken89, 2020-09-17:
Ah, then you need to provide wrappers based on the `i` module in heapless to get `const fn`s. 
Not sure how feasible that is though unfortunately.

cpfister, 2020-09-18:
Thanks to @nickray and @korken89! Now closing this issue.

closed 2020-09-18 by cpfister