rust-embedded/heapless #174: Is there a reason Send is not implemented for spsc::Producer<_,_,_,SingleCore>?

dimpolo, 2020-09-17:
I got stuck trying to put a Producer in a Mutex<RefCell<Option<Producer<_,_,_,SingleCore>>>> because Send was not implemented for it.
After switching to MultiCore everything worked.

This seems to be the code in question.
```rust
unsafe impl<'a, T, N, U> Send for Producer<'a, T, N, U>
where
    N: ArrayLength<T>,
    T: Send,
    U: sealed::Uxx,
{
}
```

korken89, 2020-09-18:
This seems weird, it should not be a problem.
Which version of heapless are you running?

The `Send` impl looks like this now and `XCore` is implemented for `SingleCore`:

```rust
unsafe impl<'a, T, N, U, C> Send for Producer<'a, T, N, U, C>
where
    N: ArrayLength<T>,
    T: Send,
    U: sealed::Uxx,
    C: sealed::XCore,
{
}
```

The important part is that the

dimpolo, 2020-09-18:
Oh I see, should have looked more carefully.
This got fixed in #149 which wasn't included in the 0.5.5 release which I was using.
Using master everything works.

closed 2020-09-18 by dimpolo

korken89, 2020-09-18:
Oh it was missed in the release, I'll fix a release right away!