rust-embedded/heapless #177: Could not compile `as-slice` when targeting AVR

ToddG, 2020-09-23:
I'm a newbie to embedded rust (and rust in general). When I add `heapless` to this project, it fails to build. If I comment out `heapless` in the `Cargo.toml`, then it builds fine.
 
## REPRO

1. Create lib

```
(base) ~/temp/break $ cargo new --lib abc
     Created library `abc` package
```

2. Build

```
(base) ~/temp/break/abc $ cargo build
   Compiling abc v0.1.0 (/home/todd/temp/break/abc)
    Finished dev [unoptimized + debuginfo] target(s) in 0.23s
```

3. Add `heapless` to Cargo.toml

`Cargo.toml`
```
[dependencies]
heapless = "0.5.6"
```

```
(base) ~/temp/break/abc $ cargo build
    Updating crates.io index
   Compiling typenum v1.12.0
   Compiling byteorder v1.3.4
   Compiling stable_deref_trait v1.2.0
   Compiling heapless v0.5.6
   Compiling hash32 v0.1.1
   Compiling generic-array v0.13.2
   Compiling generic-array v0.12.3
   Compiling as-slice v0.1.3
   Compiling abc v0.1.0 (/home/todd/temp/break/abc)
    Finished dev [unoptimized + debuginfo] target(s) in 6.80s
```

4. Switch to nightly and build

```
$ rustup override set nightly

rustup override set nightly
info: using existing install for 'nightly-x86_64-unknown-linux-gnu'
info: override toolchain for '/home/todd/temp/break/abc' set to 'nightly-x86_64-unknown-linux-gnu'

  nightly-x86_64-unknown-linux-gnu unchanged - rustc 1.48.0-nightly (dbb73f8f7 2020-09-12)

(base) ~/temp/break/abc $ cargo build
   Compiling typenum v1.12.0
   Compiling byteorder v1.3.4
   Compiling stable_deref_trait v1.2.0
   Compiling heapless v0.5.6
   Compiling hash32 v0.1.1
   Compiling generic-array v0.13.2
   Compiling generic-array v0.12.3
   Compiling as-slice v0.1.3
   Compiling abc v0.1.0 (/home/todd/temp/break/abc)
    Finished dev [unoptimized + debuginfo] target(s) in 6.78s
```

5. Add AVR stuff

```
copy from/somewhere/avr-atmega328p.json .
copy from/somewhere/.cargo/config.toml .cargo/.
```

(project files attached)

# ERROR

```(base) ~/temp/break/abc $ cargo build --verbose
    Updating crates.io index
       Fresh core v0.0.0 (/home/todd/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
       Fresh rustc-std-workspace-core v1.99.0 (/home/todd/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
       Fresh compiler_builtins v0.1.35
       Fresh typenum v1.12.0
       Fresh byteorder v1.3.4
       Fresh stable_deref_trait v1.2.0
       Fresh generic-array v0.13.2
       Fresh generic-array v0.12.3
       Fresh hash32 v0.1.1
   Compiling as-slice v0.1.3
     Running `rustc --crate-name as_slice /home/todd/.cargo/registry/src/github.com-1ecc6299db9ec823/as-slice-0.1.3/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C panic=abort -C embed-bitcode=no -C debuginfo=2 -C metadata=fee3837ddb75675c -C extra-filename=-fee3837ddb75675c --out-dir /home/todd/temp/break/abc/target/avr-atmega328p/debug/deps --target /home/todd/temp/break/abc/avr-atmega328p.json -L dependency=/home/todd/temp/break/abc/target/avr-atmega328p/debug/deps -L dependency=/home/todd/temp/break/abc/target/debug/deps --extern 'noprelude:compiler_builtins=/home/todd/temp/break/abc/target/avr-atmega328p/debug/deps/libcompiler_builtins-7d4617280997064b.rmeta' --extern 'noprelude:core=/home/todd/temp/break/abc/target/avr-atmega328p/debug/deps/libcore-2d142ffef2835bf2.rmeta' --extern generic_array=/home/todd/temp/break/abc/target/avr-atmega328p/debug/deps/libgeneric_array-15924d526938d28b.rmeta --extern ga13=/home/todd/temp/break/abc/target/avr-atmega328p/debug/deps/libgeneric_array-16c4ac167d17a181.rmeta --extern stable_deref_trait=/home/todd/temp/break/abc/target/avr-atmega328p/debug/deps/libstable_deref_trait-900fefb67ab7dd77.rmeta -Z unstable-options --cap-lints allow`
error[E0080]: evaluation of constant value failed
   --> /home/todd/.cargo/registry/src/github.com-1ecc6299db9ec823/as-slice-0.1.3/src/lib.rs:169:102
    |
169 |     250, 251, 252, 253, 254, 255, 256, 1 << 9, 1 << 10, 1 << 11, 1 << 12, 1 << 13, 1 << 14, 1 << 15, 1 << 16
    |                                                                                                      ^^^^^^^ attempt to shift left by 16_i32 which would overflow

error[E0119]: conflicting implementations of trait `AsSlice` for type `[_; 0]`:
   --> /home/todd/.cargo/registry/src/github.com-1ecc6299db9ec823/as-slice-0.1.3/src/lib.rs:138:13
    |
138 |               impl<T> AsSlice for [T; $N] {
    |               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |               |
    |               first implementation here
    |               conflicting implementation for `[_; 0]`
...
156 | / array!(
157 | |     0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25,
158 | |     26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49,
159 | |     50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73,
...   |
169 | |     250, 251, 252, 253, 254, 255, 256, 1 << 9, 1 << 10, 1 << 11, 1 << 12, 1 << 13, 1 << 14, 1 << 15, 1 << 16
170 | | );
    | |__- in this macro invocation
    |
    = note: this error originates in a macro (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0119]: conflicting implementations of trait `AsMutSlice` for type `[_; 0]`:
   --> /home/todd/.cargo/registry/src/github.com-1ecc6299db9ec823/as-slice-0.1.3/src/lib.rs:147:13
    |
147 |               impl<T> AsMutSlice for [T; $N] {
    |               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |               |
    |               first implementation here
    |               conflicting implementation for `[_; 0]`
...
156 | / array!(
157 | |     0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25,
158 | |     26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49,
159 | |     50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73,
...   |
169 | |     250, 251, 252, 253, 254, 255, 256, 1 << 9, 1 << 10, 1 << 11, 1 << 12, 1 << 13, 1 << 14, 1 << 15, 1 << 16
170 | | );
    | |__- in this macro invocation
    |
    = note: this error originates in a macro (in Nightly builds, run with -Z macro-backtrace for more info)

error: aborting due to 3 previous errors

Some errors have detailed explanations: E0080, E0119.
For more information about an error, try `rustc --explain E0080`.
error: could not compile `as-slice`

Caused by:
  process didn't exit successfully: `rustc --crate-name as_slice /home/todd/.cargo/registry/src/github.com-1ecc6299db9ec823/as-slice-0.1.3/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C panic=abort -C embed-bitcode=no -C debuginfo=2 -C metadata=fee3837ddb75675c -C extra-filename=-fee3837ddb75675c --out-dir /home/todd/temp/break/abc/target/avr-atmega328p/debug/deps --target /home/todd/temp/break/abc/avr-atmega328p.json -L dependency=/home/todd/temp/break/abc/target/avr-atmega328p/debug/deps -L dependency=/home/todd/temp/break/abc/target/debug/deps --extern 'noprelude:compiler_builtins=/home/todd/temp/break/abc/target/avr-atmega328p/debug/deps/libcompiler_builtins-7d4617280997064b.rmeta' --extern 'noprelude:core=/home/todd/temp/break/abc/target/avr-atmega328p/debug/deps/libcore-2d142ffef2835bf2.rmeta' --extern generic_array=/home/todd/temp/break/abc/target/avr-atmega328p/debug/deps/libgeneric_array-15924d526938d28b.rmeta --extern ga13=/home/todd/temp/break/abc/target/avr-atmega328p/debug/deps/libgeneric_array-16c4ac167d17a181.rmeta --extern stable_deref_trait=/home/todd/temp/break/abc/target/avr-atmega328p/debug/deps/libstable_deref_trait-900fefb67ab7dd77.rmeta -Z unstable-options --cap-lints allow` (exit code: 1)

```

ToddG, 2020-09-23:
Added project files:

[abc.zip](https://github.com/japaric/heapless/files/5268007/abc.zip)

korken89, 2020-09-23:
Hi, I have released `as-slice` v0.1.4 which should solve your issue.

ToddG, 2020-09-23:
@korken89 Thank you! Verified this fixes my issue and opened a PR:

https://github.com/japaric/heapless/pull/178

korken89, 2020-09-23:
Fixed it #178

closed 2020-09-23 by korken89