rust-embedded/heapless #181: Unsoundness in vec::IntoIter::clone()

Dirbaio, 2020-11-02:
`IntoIter`'s Clone implementation clones the entire inner Vec. If the iterator has been partially consumed, only the non-consumed items should be cloned.

The following code demonstrates the issue:

```rust
use heapless::consts::*;
use heapless::Vec;

#[derive(Debug)]
struct Foo {
    alive: bool,
    n: u32,
}

impl Foo {
    fn new(n: u32) -> Self {
        Self { n, alive: true }
    }
}

impl Clone for Foo {
    fn clone(&self) -> Self {
        if !self.alive {
            panic!("Trying to clone dead Foo #{}", self.n);
        }
        println!("cloning Foo #{}", self.n);
        Self {
            n: self.n,
            alive: self.alive,
        }
    }
}

impl Drop for Foo {
    fn drop(&mut self) {
        self.alive = false;
        println!("Dropping Foo #{}, now alive = {}", self.n, self.alive);
    }
}

fn main() {
    println!("Hello, world!");
    let mut v: Vec<Foo, U16> = Vec::new();
    v.push(Foo::new(1)).unwrap();
    v.push(Foo::new(2)).unwrap();
    v.push(Foo::new(3)).unwrap();

    let mut i = v.into_iter();

    let item = i.next();
    println!("popped {:?}", item);
    core::mem::drop(item);

    println!("cloning iter");

    let mut j = i.clone();
}
```

This outputs

```
Hello, world!
popped Some(Foo { alive: true, n: 1 })
Dropping Foo #1, now alive = false
cloning iter
cloning Foo #1
cloning Foo #2
cloning Foo #3
Dropping Foo #2, now alive = false
Dropping Foo #3, now alive = false
Dropping Foo #2, now alive = false
Dropping Foo #3, now alive = false
```

After `Foo 1` is popped from the iterator, cloning the iterator will still clone the "dead" `Foo 1`. Only `Foo 2` and `Foo 3` should be cloned.

Shnatsel, 2021-02-27:
Heads up: this issue has been included in the [RustSec advisory database](https://github.com/RustSec/advisory-db). It will be surfaced by tools such as [cargo-audit](https://github.com/RustSec/cargo-audit) or [cargo-deny](https://github.com/EmbarkStudios/cargo-deny) from now on.

Once a fix is released to crates.io, please open a pull request to update the advisory with the patched version, or file an issue on [the advisory database repository](https://github.com/RustSec/advisory-db).

korken89, 2021-03-02:
Fixed in #195

korken89, 2021-03-02:
Now released as 0.6.1

korken89, 2021-03-02:
And the PR is merged: https://github.com/RustSec/advisory-db/pull/791/files

closed 2021-03-02 by korken89