rust-embedded/heapless #183: Unable to add a generic Vec as struct field

Maldus512, 2020-11-11:
Hello,
I'm trying to create a custom struct type containing an `heapless::Vec` generic on its size. Unfortunately the compiler complains that the size type does not implement `ArrayLength`, even though I specified the constraint:

```
use generic_array::ArrayLength;
use heapless::Vec;

struct MyStruct<T, N: ArrayLength<T>> {
    vector: Vec<T, N>,
}
```

```
error[E0277]: the trait bound `N: generic_array::ArrayLength<T>` is not satisfied
   --> src/libs/throttler.rs:5:5
    |
5   |     vector: Vec<T, N>,
    |     ^^^^^^^^^^^^^^^^^ the trait `generic_array::ArrayLength<T>` is not implemented for `N`
    | 
   ::: /home/maldus/.cargo/registry/src/github.com-1ecc6299db9ec823/heapless-0.5.5/src/vec.rs:165:8
    |
165 |     N: ArrayLength<T>;
    |        -------------- required by this bound in `heapless::vec::Vec`
    |
help: consider further restricting this bound
    |
4   | struct MyStruct<T, N: ArrayLength<T> + generic_array::ArrayLength<T>> {
    |                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

Any idea why this happens? The suggested help is quite nonsensical because it just duplicates the constraint already present.
Interestingly, everything works if I use `GenericArray` instead:

```
use generic_array::{ArrayLength, GenericArray};

struct MyStruct<T, N: ArrayLength<T>> {
    vector: GenericArray<T, N>,
}
```

birkenfeld, 2020-11-11:
Can you check if your `generic_array` crate is the same version as the one used by `heapless`? It may be that heapless uses a different incompatible version, in which case both will be compiled in, and the traits will be considered distinct.

(I think the error message should mention this in newer Rust compilers, but I don't know which rustc version you're using.)

Maldus512, 2020-11-11:
Spot on. I was using `generic_array = "0.14.4"`, moving back to version `0.13.0` did the trick. Thank you!

closed 2020-11-11 by Maldus512