rust-embedded/heapless #191: Compatibility issue with using heapless with AVR?

kkevlar, 2021-02-09:
Hi all, real quick question here.

I'm getting some very unexpected behavior with my toy project to test out building a embedded application with Rust.

However, the undefined behavior wouldn't be that interesting to report/debug if the authors don't expect `heapless` to be compatible with [avr-hal](https://github.com/Rahix/avr-hal).

Thank you for your help- working on getting a extremely small case where I still see this bug.

Thanks again,

Kevin

korken89, 2021-02-09:
All should work, except the features which are gated. What issue are you seeing?

kkevlar, 2021-02-11:
I'll make a new issues this week when I've made a small reproducible example. Thanks so much for your help

closed 2021-02-11 by kkevlar