rust-embedded/heapless #210: spsc: copy-less API

jordens, 2021-06-17:
As pointed out in https://github.com/japaric/heapless/blob/3b4983683d8d11a70db26b0c2678082f2d5c8a2b/src/spsc.rs#L81-L82
both `enqueue()` and `dequeue()` perform copies. If the items have significant size, this leads to a large unnecessary overhead that could be removed if the user was allowed direct &/&mut access to the item to be en/dequeued.

At the dequeue side, this can be worked around with `peek()` which gives `Option<&T>`. Yet to remove the item from the queue, one still has to `dequeue()` which may (if not optimized away by the compiler) still do the copy. 
Something similar to `peek()` for the enqueue side (maybe `poke()` plus `commit()` to advance the pointer) would be useful.
Alternatively a closure based API (`enqueue_with(f: FnOnce(&mut T) -> R) -> Result<R>` plus matching for the `dequeue()` side would solve this.
The fact that both approaches could give a &mut to uninitialized memory looks like it might be a problem though.

Is there a way to give true copy-less access to queue items at both ends or is this conceptually unlikely to be possible?

korken89, 2021-06-20:
As you come to the conclusion, usually one needs to somehow work with uninitialized memory.
This is fine though as long as one knows what they're doing.

However, my recommendation is to use a fat pointer if this kind of speed is needed.
E.g. to place the types in a `pool`. This is my personal goto approach.

jordens, 2021-06-21:
Nice. That looks very useful. Feel free to close.

closed 2021-06-21 by korken89