rust-embedded/heapless #212: `Box::freeze` allows obtaining uninitialized instances of arbitrary types

jonas-schievink, 2021-06-25:
```rust
use std::num::NonZeroU32;

use heapless::pool;
use heapless::pool::singleton::Pool;

#[derive(Debug)]
pub struct S {
    v: NonZeroU32,
}

impl AsRef<[u8]> for S {
    fn as_ref(&self) -> &[u8] {
        &[]
    }
}

pool!(P: S);

fn main() {
    static mut BUFFER: [u8; 1024] = [0; 1024];

    P::grow(unsafe { &mut BUFFER });

    let boks = P::alloc().unwrap().freeze();
    println!("{:?}", boks);
}
```

This prints `S { v: 0 }`, which cannot exist.

korken89, 2021-06-30:
Hmm, I am not sure what is the old purpose of the `freeze` method.
It could be a remnant from the pre-`MaybeUninit` days.

I'm all for just deleting it. Thoughts?

korken89, 2021-07-01:
I'll deprecate it for now and add a warning.
So it can be removed during next "major" release.

PR #217 (2021-07-01): Deprecate `Box::freeze`, prepare v0.7.3
Closes #212
closed 2021-07-01 by korken89