rust-embedded/heapless #214: Queue with capacity 1 is both empty and full

jonas-schievink, 2021-06-29:
```rust
use heapless::spsc::Queue;

fn main() {
    let queue: Queue<[u8; 16], 1> = Queue::new();
    assert!(queue.is_empty());
    assert!(!queue.is_full());
}
```

```
thread 'main' panicked at 'assertion failed: !queue.is_full()', src/main.rs:6:5
```

birkenfeld, 2021-06-29:
From the docs: "A statically allocated single producer single consumer queue with a capacity of N - 1 elements".  This is, I think, because the needed const generic math to allocate `N + 1` items is not stable yet.

Is it possible to statically assert `N > 1`?

jonas-schievink, 2021-06-29:
Oh, I see, I missed that

birkenfeld, 2021-06-29:
It's new in 0.7 (the first const generics based version).

burrbull, 2021-06-30:
> Is it possible to statically assert `N > 1`?

In nightly only.

korken89, 2021-06-30:
Hi, it's possible to do asserts in const with some tricks.
https://github.com/nvzqz/static-assertions-rs/issues/40#issuecomment-846228355

I'll look into adding them everywhere.

korken89, 2021-06-30:
Yeah, the interface is a bit unfortunate.
I hope we get more const power soon so we can get the old interface back.

PR #215 (2021-06-30): Preparing v0.7.2
Closes #214
closed 2021-06-30 by korken89