rust-embedded/heapless #216: evaluation of constant value failed

MathiasKoch, 2021-06-30:
```
error[E0080]: evaluation of constant value failed
  --> .../github.com-1ecc6299db9ec823/heapless-0.7.2/src/sealed.rs:56:32
   |
56 |     pub const GREATER: usize = L - R - 1;
   |                                ^^^^^^^^^ attempt to compute `0_usize - 1_usize`, which would overflow

error: aborting due to previous error; 7 warnings emitted

```

korken89, 2021-06-30:
This is the new const assert failing.
Either you have a Queue with N == 0 or 1 or a Vec with N == 0.
This will cause the data structures to misbehave add it's below the minimum sizes of each.

I added these checks to highlight this issue to users.

korken89, 2021-06-30:
Also note the new size definition for Queue, size = N - 1 until we get more const powers and get back the old interface

MathiasKoch, 2021-07-01:
Found it.. Had a `String<0>` in my code. Thanks for the assistance!

closed 2021-07-01 by MathiasKoch