rust-embedded/heapless #218: Is their room for an MPSC?

huntc, 2021-07-02:
Having spent some time on creating a PR for Embassy that introduces a non-alloc MPSC, there was a question posed as to whether the MPSC would be a better fit here.

The MPSC’s implementation is designed for single cores given its use of critical sections, but the API doesn’t restrict this. A multi core implementation could also be attained. 

I realise that there is a MPMC here already, but perhaps the potential for optimisations of a single consumer may be of interest. 

Here’s my Embassy PR: https://github.com/embassy-rs/embassy/pull/226, which remains open at this point.

Thanks!

korken89, 2021-07-03:
I think this would be a good fit for `heapless` :)
I have not had time to check the implementation, but it makes sense to have here.

korken89, 2023-01-10:
Hi, did you decide if you wanted to look into adding an MPSC here?

huntc, 2023-01-10:
My apologies for leaving this one open. I ultimately decided to implement it in embassy-rs where it now lives as an MPMC. Please feel free to close this if you’d like.

closed 2023-01-13 by korken89