rust-embedded/heapless #222: Support assuming initialization for pool::Box

ryan-summers, 2021-07-22:
In cases where a Box contains a large value (e.g. [u8; 1024]`, the copy can be expensive to initialize the `Box` to a defined value. 

In trivial cases like this, where only a buffer is stored, there's no need to necessary overwrite the array if the data will be written later.

What are potential downsides on adding a new function to `Box<T, Uninit>`? It seems to me like we would avoid the overhead of initializing the box when the contents are dont-care. Obviously this is quite unsafe for most of Rust's types, but is perfectly safe for generic u8s.
```rust
unsafe pub fn assume_init(self) -> Box<T, Init>
```?

korken89, 2021-09-24:
Hi,

I did some testing and when using `MaybeUninit::uninit()` it seems to optimize as expected.
Do you have a reproducible example I can use to debug this?

jordens, 2022-02-10:
I think the important thing is that the array elements need to be `MaybeUninit` and the the array itself. Can probably be closed.

```rust
#![no_std]
#![no_main]

use core::mem::MaybeUninit;

use cortex_m::asm;
use cortex_m_rt::entry;
use heapless::{
    pool,
    pool::{
        singleton::{Box, Pool},
        Uninit,
    },
};
use panic_halt as _;

const N: usize = 1024;
pool!(GOOD: [MaybeUninit<i32>; N]);
pool!(BAD: MaybeUninit<[i32; N]>);

#[entry]
fn main() -> ! {
    static mut MEM1: [u8; N] = [0; N];
    static mut MEM2: [u8; N] = [0; N];
    GOOD::grow(MEM1);
    BAD::grow(MEM2);
    test_good(GOOD::alloc().unwrap());
    test_bad(BAD::alloc().unwrap());
    loop {
        asm::nop();
    }
}

#[inline(never)]
fn test_good(b: Box<GOOD, Uninit>) {
    // let mut buf = b.init(unsafe { MaybeUninit::uninit().assume_init() });
    let mut buf = b.init([MaybeUninit::uninit(); N]);
    for (i, b) in buf.iter_mut().enumerate() {
        b.write(i as _);
    }
    drop(buf);
}

#[inline(never)]
fn test_bad(b: Box<BAD, Uninit>) {
    let mut buf = unsafe { b.init(MaybeUninit::uninit()).assume_init() };
    for (i, b) in buf.iter_mut().enumerate() {
        *b = i as _;
    }
}
```

japaric, 2022-05-10:
both `MaybeUninit<[u8; 1024]>` and `[MaybeUninit<u8>; 1024]` appear to optimize equally well with rustc v1.60

``` rust
#![no_std]
#![no_main]

use core::{mem::MaybeUninit, panic::PanicInfo};

// heapless = "0.7.10"
use heapless::pool::{Box, Init, Uninit};

#[no_mangle]
fn _start() {
    unsafe {
        (&(good as usize) as *const usize).read_volatile();
        (&(also_good as usize) as *const usize).read_volatile();
    }
}

#[no_mangle]
fn good(boxed: Box<MaybeUninit<[u8; 1024]>, Uninit>) -> Box<MaybeUninit<[u8; 1024]>, Init> {
    boxed.init(MaybeUninit::uninit())
}

#[no_mangle]
fn also_good(boxed: Box<[MaybeUninit<u8>; 1024], Uninit>) -> Box<[MaybeUninit<u8>; 1024], Init> {
    boxed.init([MaybeUninit::uninit(); 1024])
}

#[panic_handler]
fn panic(_: &PanicInfo) -> ! {
    loop {}
}
```

machine code:

``` armasm
000200e4 <_start>:
   200e4: 82 b0         sub     sp, #8
   200e6: 40 f2 03 10   movw    r0, #259
   200ea: c0 f2 02 00   movt    r0, #2
   200ee: 00 90         str     r0, [sp]
   200f0: 00 98         ldr     r0, [sp]
   200f2: 40 f2 05 10   movw    r0, #261
   200f6: c0 f2 02 00   movt    r0, #2
   200fa: 01 90         str     r0, [sp, #4]
   200fc: 01 98         ldr     r0, [sp, #4]
   200fe: 02 b0         add     sp, #8
   20100: 70 47         bx      lr

00020102 <good>:
   20102: 70 47         bx      lr

00020104 <bad>:
   20104: 70 47         bx      lr
```

so I'll close this issue. if someone runs into an optimization issue please open a new issue with steps to reproduce

related: I think the `ObjectPool` API proposed in RFC #286 would serve better the use case of byte buffers

closed 2022-05-10 by japaric

jordens, 2022-05-10:
At least with the example above, I can still see the additional `memcpy` in `test_bad` in the example above with rust 1.60.
Does it matter that is about [`heapless::pool::singleton::Box`](https://github.com/japaric/heapless/blob/0cb35c717069364b27a7a4ea5b7ad2ecdd52f89f/src/pool/singleton.rs#L94)  not [`heapless::pool::Box`](https://github.com/japaric/heapless/blob/0cb35c717069364b27a7a4ea5b7ad2ecdd52f89f/src/pool/mod.rs#L441) ?

But in any case: the ideas from #286 sound great!