rust-embedded/heapless #248: Feature gate of `spsc` not working? (Compile error)

piegamesde, 2021-10-05:
Related to #179 and others. I'm trying to compile the crate for `riscv32im-unknown-none-elf`, which has no atomics (or friends). I get the following compile error:

```
error[E0432]: unresolved import `core::sync::atomic::AtomicUsize`
  --> $CARGO_HOME/registry/src/github.com-1ecc6299db9ec823/heapless-0.7.7/src/spsc.rs:92:20
   |
92 |     sync::atomic::{AtomicUsize, Ordering},
   |                    ^^^^^^^^^^^ no `AtomicUsize` in `sync::atomic`
```

This is intriguing, because the code already has the `spsc` behind a compile flag:

```rust
#[cfg(has_atomics)]
pub mod spsc;
```

But why is it not working? Since it appeared to work on `riscv32i-unknown-none-elf` for some reason, here's my custom target file that I'm using. It may be the cause of the error:

```json
{
  "arch": "riscv32",
  "atomic-cas": false,
  "cpu": "generic-rv32",
  "data-layout": "e-m:e-p:32:32-i64:64-n32-S128",
  "eh-frame-header": false,
  "emit-debug-gdb-scripts": false,
  "executables": true,
  "features": "+m",
  "is-builtin": false,
  "linker": "rust-lld",
  "linker-flavor": "ld.lld",
  "llvm-target": "riscv32",
  "max-atomic-width": 0,
  "panic-strategy": "abort",
  "relocation-model": "static",
  "target-pointer-width": "32"
}
```

korken89, 2021-10-06:
Hi,

We feature gate based on the available targets in `rustup`.
From what I can see `riscv32im-unknown-none-elf` does not exist as a target?

```console
> rustup target list | grep riscv
riscv32i-unknown-none-elf
riscv32imac-unknown-none-elf
riscv32imc-unknown-none-elf
riscv64gc-unknown-linux-gnu
riscv64gc-unknown-none-elf
riscv64imac-unknown-none-elf
```

However to extend with more targets is easily done in `build.rs` where we have a list of targets that does not have CAS or atomics.
If it's not an official target you can fork this repository and add you target.

closed 2021-10-06 by korken89

piegamesde, 2021-10-06:
> We feature gate based on the available targets in rustup.

Please don't; this is a custom target. Especially on RISCV, there arbitrarily many different targets due to their mix-and-match modular design. Ideally, this would be a property of the target you could simply read and use, but I assume you'd be already using it if there was.

As I don't need this feature anyways, so maybe add a library-level feature gate like done for `cas`?

piegamesde, 2021-10-07:
I did some research and apparently the best way to solve this is to wait for `#[cfg(target_has_atomic)]` to be stabilized and then use that instead.