rust-embedded/heapless #250: Does not compile when rustflags="-C", "link-arg=-nostdlib"

AsafFisher, 2021-11-10:
```
error: failed to run custom build command for `heapless v0.7.7`

Caused by:
  process didn't exit successfully: `/mnt/c/Users/asaff/Documents/Dev/Github/tool/target/debug/build/heapless-629176dd3fec225f/build-script-build` (signal: 11, SIGSEGV: invalid memory reference)
```

TDHolmes, 2022-01-14:
Works fine for me. I would try again. Anyways based on the error, not sure this would be a `heapless` bug anyways.

japaric, 2022-05-10:
looks like the build script was built / linked with `-nostdlib`. make sure you pass the `--target` flag to Cargo even if you are doing native compilation (annoyingly this matters), e.g. `cargo build --target x86_64-unknown-linux-gnu` on Linux. you can get the target triple from the `host` line in the output of `rustc -Vv`

closing as this does not seem like a heapless bug. feel free to open a new issue with steps to reproduce if you still have problems

closed 2022-05-10 by japaric