rust-embedded/heapless #252: Cannot declare 0-length capacity Vector

haata, 2021-11-12:
This is a bit of an odd case so I'm not sure if you want to handle it.

I'll give the example first, and then my reasoning behind it (also for anyone else who runs into this).

With heapless 0.7.8 this will not compile.
```rust
use heapless::Vec;

pub struct MyStruct<const H: usize> {
    pub data: Vec<u8, H>,
}

fn main() {
    let _val = MyStruct::<0> {
        data: Vec::new(),
    };

}
```

```bash
   Compiling rusttest v0.1.0 (/home/hyatt/git/rusttest)
error[E0080]: evaluation of `heapless::sealed::Assert::<0_usize, 0_usize>::GREATER` failed
  --> /home/hyatt/.cargo/registry/src/github.com-1ecc6299db9ec823/heapless-0.7.8/src/sealed.rs:93:32
   |
93 |     pub const GREATER: usize = L - R - 1;
   |                                ^^^^^^^^^ attempt to compute `0_usize - 1_usize`, which would overflow

note: the above error was encountered while instantiating `fn heapless::sealed::greater_than_0::<0_usize>`
  --> /home/hyatt/.cargo/registry/src/github.com-1ecc6299db9ec823/heapless-0.7.8/src/vec.rs:57:9
   |
57 |         crate::sealed::greater_than_0::<N>();
   |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

For more information about this error, try `rustc --explain E0080`.
error: could not compile `rusttest` due to previous error
```

I had been using 0 capacity Vectors for a specific case for an RPC mechanism I'm building where the data field is a heapless::Vec that defines the maximum capacity of the data field (which can vary in a few cases). For 1 packet type the data field is always 0 length so I have set the Vector as 0 length.

This won't have any effect on the wire-format, so I'm not terribly concerned. But it took me a while to isolate the problem as the assert does not show the original code line.

(also the heapless crate was the sole reason for my jump from C to Rust for embedded development, it's excellent! thank you!)

korken89, 2021-12-16:
If you can have a look at the implementations and make sure that a 0 size is fine, I see no problem in allowing it.
I don't have so much bandwidth now, feel free to make a PR. This is a relaxation of a bound so it would not be a breaking change if it is fine.

haata, 2021-12-28:
I've pushed a PR /w a test.
So far I don't see any issues (I was using zero-length Vecs with heapless before the switch to const generics without any issues).

haata, 2022-01-10:
Thanks for merging!

closed 2022-01-10 by haata