rust-embedded/heapless #258: IndexMap doesn't allow update of existing record when full

jamessewell, 2021-12-16:
The docs seem to indicate this should be possible:

> If an equivalent key already exists in the map: the key remains and retains in its place in the order, its corresponding value is updated with value and the older value is returned inside Some(_).

However this gets an error on the final `unwrap()`

```
use heapless::FnvIndexMap;

fn main() {
    let mut map = FnvIndexMap::<_, _, 4>::new();

    map.insert(1, 1).unwrap();
    map.insert(2, 1).unwrap();
    map.insert(3, 1).unwrap();

    // test updates work when not full
    map.insert(3, 2).unwrap();

    map.insert(4, 1).unwrap();

    // It's full
    assert_eq!(map.len(), map.capacity());

    // Update an existing value -> this errors
    map.insert(3, 3).unwrap();
}
```

jamessewell, 2021-12-16:
An actually looking at the code this is exactly right:

```
    pub fn insert(&mut self, key: K, value: V) -> Result<Option<V>, (K, V)> {
        if self.core.entries.is_full() {
            Err((key, value))
```

Given that you can test for the key being in the IndexMap can this be weakened?

korken89, 2021-12-16:
It can probably be weakened, but I'm not sure how.
I'll ping @japaric on this. :)

japaric, 2021-12-16:
I agree that updating an entry should be possible even when the map is full. In the current implementation, the "is full" check is too eager and instead should be done later in the logic.

`heapless::IndexMap` is based on an old version of the `indexmap`; this particular feature seems easier to implement in the latest version of `indexmap` which has been refactored quite a bit.

https://github.com/bluss/indexmap/blob/0af3c466a7c69ff27bfd0024562260fde171d1be/src/map/core.rs#L233-L241

In the linked code, I think doing the 'is full' check in the None branch would get us the desired behavior.