rust-embedded/heapless #278: the trait `IntoWasmAbi` is not implemented for `heapless::Vec<u8, 8_usize>`

brandonros, 2022-04-20:
I'm in no man's land, lol

```
Brandons-MacBook-Air:protocol brandonros 2022-04-19 23:59:14 $ wasm-pack build --target web
[INFO]: 🎯  Checking for the Wasm target...
[INFO]: 🌀  Compiling to Wasm...
   Compiling protocol v0.1.0 (/Users/brandonros/Desktop/stm_j2534/protocol)
error[E0277]: the trait bound `heapless::Vec<u8, 8_usize>: IntoWasmAbi` is not satisfied
  --> src/lib.rs:29:1
   |
29 | #[wasm_bindgen]
   | ^^^^^^^^^^^^^^^ the trait `IntoWasmAbi` is not implemented for `heapless::Vec<u8, 8_usize>`
   |
   = note: this error originates in the attribute macro `wasm_bindgen` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0277]: the trait bound `heapless::Vec<u8, 8_usize>: FromWasmAbi` is not satisfied
  --> src/lib.rs:29:1
   |
29 | #[wasm_bindgen]
   | ^^^^^^^^^^^^^^^ the trait `FromWasmAbi` is not implemented for `heapless::Vec<u8, 8_usize>`
   |
   = note: this error originates in the attribute macro `wasm_bindgen` (in Nightly builds, run with -Z macro-backtrace for more info)

For more information about this error, try `rustc --explain E0277`.
error: could not compile `protocol` due to 2 previous errors
Error: Compiling your crate to WebAssembly failed
Caused by: failed to execute `cargo build`: exited with exit status: 101
  full command: "cargo" "build" "--lib" "--release" "--target" "wasm32-unknown-unknown"
Brandons-MacBook-Air:protocol brandonros 2022-04-19 23:59:30 $ 
```

```rust
#[derive(Debug, Serialize, Deserialize)]
#[wasm_bindgen]
pub struct CanFrame {
    pub id: u32,
    pub data: Vec<u8, 8>,
}
```

japaric, 2022-05-12:
you can use a newtype to implement the trait, e.g.
``` rust
struct MyVec<T, const N: usize>(Vec<T, N>);

impl<T, const N: usize> IntoWasmAbi for MyVec<T, N> {
    // ..
}

#[wasm_bindgen]
pub struct CanFrame {
    pub id: u32,
    pub data: MyVec<u8, 8>,
}
```

brandonros, 2022-05-12:
I was compiling Darwin target on accident instead of the arm thumb thingie, thank you! Sorry!

closed 2022-05-12 by brandonros

mimi89999, 2025-04-01:
Is there a ready implementation that can be used?