rust-embedded/heapless #279: bloated binary while using `f32` as Queue data type

pm-daniel, 2022-04-24:
Hi, 
I have no idea how different data types are implemented for the `Queue` struct but something seems wrong
just tested output binaries with `cargo-bloat` one is using `u32` and another is using `f32` as `Queue` data type 
here are the results:

`u32`
```
File  .text    Size          Crate Name
0.7%  18.6%  3.9KiB      [Unknown] USB_HP_CAN_TX
0.4%  11.5%  2.4KiB DiPho_firmware DiPho_firmware::app::rtic_ext::main::__rtic_init_resources
0.2%   4.3%    924B            std compiler_builtins::int::specialized_div_rem::u64_div_rem
0.1%   4.0%    846B      [Unknown] memmove
0.1%   3.6%    762B      [Unknown] CAN_SCE
0.1%   3.5%    750B            std core::str::slice_error_fail
0.1%   3.5%    748B            std <core::fmt::builders::PadAdapter as core::fmt::Write>::write_str
0.1%   3.0%    634B    stm32_usbd? <stm32_usbd::bus::UsbBus<USB> as usb_device::bus::UsbBus>::alloc_ep
0.1%   2.6%    560B     usb_device usb_device::endpoint::Endpoint<B,usb_device::endpoint::In>::write
0.1%   2.5%    542B     usb_device usb_device::endpoint::Endpoint<B,usb_device::endpoint::Out>::read
0.1%   2.4%    502B            std core::fmt::Formatter::pad_integral
0.1%   2.3%    490B   usbd_serial? <usbd_serial::serial_port::SerialPort<B,RS,WS> as usb_device::class::UsbClass<B>>::get_configu...
0.1%   2.1%    450B    usbd_serial usbd_serial::serial_port::SerialPort<B>::new
0.1%   1.9%    398B      [Unknown] SysTick
0.1%   1.8%    388B            std core::fmt::write
0.1%   1.8%    382B         fugit? <fugit::instant::Instant<T,_,_> as core::fmt::Debug>::fmt
0.1%   1.6%    332B      [Unknown] memcpy
0.1%   1.5%    326B            std <&T as core::fmt::Debug>::fmt
0.1%   1.5%    324B      [Unknown] TIM2
0.0%   1.2%    264B            std core::fmt::num::imp::<impl core::fmt::Display for u64>::fmt
0.9%  24.5%  5.1KiB                And 75 smaller methods. Use -n N to show more.
3.6% 100.0% 20.8KiB                .text section size, the file size is 581.2KiB
```

`f32`
```
File  .text    Size          Crate Name
1.3%  19.2%  9.4KiB            std core::fmt::float::float_to_decimal_common_exact
1.3%  18.3%  8.9KiB            std core::num::flt2dec::strategy::dragon::format_shortest
0.5%   7.9%  3.9KiB      [Unknown] USB_HP_CAN_TX
0.5%   6.9%  3.4KiB            std core::num::flt2dec::strategy::dragon::mul_pow10
0.4%   5.4%  2.6KiB            std core::num::flt2dec::strategy::grisu::format_shortest_opt
0.3%   4.9%  2.4KiB DiPho_firmware DiPho_firmware::app::rtic_ext::main::__rtic_init_resources
0.1%   1.8%    924B            std compiler_builtins::int::specialized_div_rem::u64_div_rem
0.1%   1.7%    846B      [Unknown] memmove
0.1%   1.5%    762B      [Unknown] CAN_SCE
0.1%   1.5%    750B            std core::str::slice_error_fail
0.1%   1.5%    748B            std <core::fmt::builders::PadAdapter as core::fmt::Write>::write_str
0.1%   1.4%    690B            std core::fmt::Formatter::write_formatted_parts
0.1%   1.3%    634B    stm32_usbd? <stm32_usbd::bus::UsbBus<USB> as usb_device::bus::UsbBus>::alloc_ep
0.1%   1.2%    614B            std core::num::bignum::Big32x40::mul_pow2
0.1%   1.2%    590B            std core::fmt::float::float_to_exponential_common_shortest
0.1%   1.1%    572B            std <&T as core::fmt::Debug>::fmt
0.1%   1.1%    560B     usb_device usb_device::endpoint::Endpoint<B,usb_device::endpoint::In>::write
0.1%   1.1%    542B     usb_device usb_device::endpoint::Endpoint<B,usb_device::endpoint::Out>::read
0.1%   1.0%    502B            std core::fmt::Formatter::pad_integral
0.1%   1.0%    490B   usbd_serial? <usbd_serial::serial_port::SerialPort<B,RS,WS> as usb_device::class::UsbClass<B>>::get_configu...
1.3%  18.8%  9.2KiB                And 89 smaller methods. Use -n N to show more.
6.9% 100.0% 48.8KiB                .text section size, the file size is 706.3KiB
```
why is that happening?

korken89, 2022-04-24:
Most likely you have an unwrap or expect on an type that has the f32 in it causing it to pull in all the formatting bloat associated to it.

japaric, 2022-05-12:
I concur with korken. perhaps you have something like `my_queue.enqueue(1.2f32).unwrap()`; `enqueue` returns `f32` as the error type so that will add a panicking branch that formats the `f32`. you can avoid formatting the `f32` using `ok` on the `Result` to drop the error type and turn it into an `Option`, i.e. `my_queue.enqueue(1.2f32).ok().unwrap()`

Feel free to reopen if you have an example that shows a panicking branch in the implementation of a `Queue` method, e.g. `let _ = my_queue.enqueue(val);`. I checked the machine code for lack of panicking branches long time ago but if any has crept in, it'd be good to remove it.

closed 2022-05-12 by japaric

TDHolmes, 2022-10-11:
What architecture are you using? Does it have hardware float?