rust-embedded/heapless #312: AVR Applications using the heapless crate fail to compile

mutantbob, 2022-08-24:
If you create an AVR project using `cargo generate --git https://github.com/Rahix/avr-hal-template.git` and add a dependency on heapless (0.7.16) compilation triggers an error

```
error: cannot find macro `llvm_asm` in this scope
   --> /home/thoth/.cargo/registry/src/github.com-1ecc6299db9ec823/critical-section-0.2.7/src/lib.rs:127:13
    |
127 |             llvm_asm!(
    |             ^^^^^^^^
```

This is because heapless depends on atomic-polyfill v0.1.10 which depends on critical-section v0.2.7 which uses the obsolete `llvm_asm!` macro.

Upgrading heapless to use atomic-polyfill 1.0.0 or later resolves this issue.

booo, 2022-11-16:
As far as I can see this was solved in b0e4cb8af01a4b85bab3ab4497e05ebc30d81d99.

I would appreciate a new release with this change so we can start integrating it in other projects.

rihardsk, 2023-03-31:
Hey, sorry to bother, but is there anything specific preventing a new release being created? It would be great to have this fixed so that downstream projects can benefit.

newAM, 2023-10-31:
This should be fixed on the `main` branch, feel free to re-open if not.

This isn't in a release yet, but we're working on that, tracking issue #373

closed 2023-10-31 by newAM