rust-embedded/heapless #321: What "order" is it for IndexSet ? appears to be insertion order

correabuscar, 2022-10-19:
The doc says:
`Return an iterator over the values of the set, in their order`
and
`Will print in an arbitrary order.`
from:
https://github.com/japaric/heapless/blob/df890b1a4bc512ef6e8eb41165861a46a9c35b5a/src/indexset.rs#L114-L128

But `in their order` and `arbitrary order` are somewhat confusing, especially since the books [example](https://github.com/japaric/heapless/blob/df890b1a4bc512ef6e8eb41165861a46a9c35b5a/src/indexset.rs#L62-L80) appears to return the items in insertion order. (and my test below confirms it)

Can this order be clarified in the docs, please? is it really `arbitrary order` or is it insertion order? 
 and is the order the same for `.iter()` as it's for `for book in &books` (this I might know if I weren't a Rust newbie, ie. maybe that `for` is secretly just calling `books.iter()` ?)

My test shows that insertion order is kept, but maybe that's just a coincidence!!?:
1. `cd /tmp`
2. `cargo new --lib wtw`
3. `cd wtw`
4. `echo 'heapless = "0.7.9"' >> Cargo.toml`
5. add the following test into `src/lib.rs`:
```rust
#[test]
    fn kept_insertion_order_ints() {
        use heapless::FnvIndexSet;
        const LAST: usize = 128; //must be power of two, FnvIndexSet requirement
        type Numbers = u32;
        let mut nums = FnvIndexSet::<Numbers, LAST>::new();
        const FIRST:Numbers=1;
        const THROWER:Numbers=1000;
        const PICKER:Numbers=100;
        for i in FIRST..=LAST as Numbers {
            let r: Result<bool, Numbers>;
            r=if i==PICKER {
                nums.insert(THROWER)
            } else {
                nums.insert(i)
            };
            //println!("{r:?}");
            assert_eq!(r,Ok(true));
        }
        let mut i:Numbers = FIRST;
        let mut iter=nums.iter();
        for j in &nums {

            if let Some(cur)=iter.next() {
                println!("{:?}", cur);
                if i==PICKER {
                    assert_eq!(*cur,THROWER);
                    assert_eq!(*j, THROWER);
                } else {
                    assert_eq!(*cur,i);
                    assert_eq!(*j, i);
                }
            }

            i += 1;
        }
    }
```

6. `cargo test -- --show-output`

```
1
2
3
4
...
96
97
98
99
1000
101
102
103
104
105
...
125
126
127
128
```

correabuscar, 2022-10-20:
I see that I was wrong and it is indeed `arbitrary order`:
```rust
#[test]
    fn kept_insertion_order_ints() {
        use heapless::FnvIndexSet;
        const LAST: usize = 128; //must be power of two, FnvIndexSet requirement
        type Numbers = u32;
        let mut nums = FnvIndexSet::<Numbers, LAST>::new();
        const FIRST:Numbers=1;
        const THROWER:Numbers=1000;
        use std::ops::RangeInclusive;
        //const DELETER:RangeInclusive<Numbers>=(FIRST+102)..=(LAST as Numbers - 5);
        const DELETER:[RangeInclusive<Numbers>; 3]=[
            (FIRST+10)..=(FIRST + 25),
            (FIRST+50)..=(FIRST + 85),
            (FIRST+102)..=(LAST as Numbers -5),
        ];
        const PICKER:Numbers=100;
        for i in FIRST..=LAST as Numbers {
            let r: Result<bool, Numbers>;
            r=if i==PICKER {
                nums.insert(THROWER)
            } else {
                nums.insert(i)
            };
            //println!("{r:?}");
            assert_eq!(r,Ok(true));

            //having removal done here doesn't change order of existing items:
            /*for all_d in DELETER {
                if all_d.contains(&i) {
                    let b=nums.remove(&i);
                    assert_eq!(b,true);
                }
            }*/
        }

        //but having removal done here, does change the order of existing items:
        for i in FIRST..=LAST as Numbers {
            for all_d in DELETER {
                if all_d.contains(&i) {
                    let b=nums.remove(&i);
                    assert_eq!(b,true);
                }
            }
        }

        for j in &nums {
            println!("now: {j}");
        }

        let mut i:Numbers = FIRST;
        let mut iter=nums.iter();
        for j in &nums {

            if let Some(cur)=iter.next() {
                println!("{:?}", cur);
                if i==PICKER {
                    assert_eq!(*cur,THROWER);
                    assert_eq!(*j, THROWER);
                } else {
                    for all_d in DELETER {
                        while all_d.contains(&i) {
                            i+=1;
                        }
                    }
                        assert_eq!(*cur,i);
                        assert_eq!(*j, i);
                }
            }

            i += 1;
        }
    }
```
```
now: 1
now: 2
now: 3
now: 4
now: 5
now: 6
now: 7
now: 8
now: 9
now: 10
now: 128
now: 127
now: 126
now: 125
now: 124
now: 91
now: 90
now: 89
now: 88
now: 87
now: 102
now: 101
now: 1000
now: 99
now: 98
now: 97
now: 27
now: 28
now: 29
now: 30
now: 31
now: 32
now: 33
now: 34
now: 35
now: 36
now: 37
now: 38
now: 39
now: 40
now: 41
now: 42
now: 43
now: 44
now: 45
now: 46
now: 47
now: 48
now: 49
now: 50
now: 96
now: 95
now: 94
now: 93
now: 92
1
2
3
4
5
6
7
8
9
10
128
thread 'tests::kept_insertion_order_ints' panicked at 'assertion failed: `(left == right)`
  left: `128`,
 right: `27`', fifo_hashset/src/lib.rs:118:25
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
test tests::kept_insertion_order_ints ... FAILED

```

japaric, 2022-10-20:
I agree that the documentation could be clearer. instead of "in their order", it should say "in insertion order"

the reason you are not seeing "insertion order" in your second example is that the `remove` API breaks the "insertion order". if you want to remove and preserve the insertion order then you should use `shift_remove`, which we have not yet ported from [upstream](https://docs.rs/indexmap/1.9.1/indexmap/set/struct.IndexSet.html#method.shift_remove).

commit fdcfd4c2 (2022-10-20): clarify the iteration order of IndexMap / IndexSet

closes #321
commit 8dcee5c7 (2022-10-20): Merge #322

322: clarify the iteration order of IndexMap / IndexSet r=japaric a=japaric

closes #321
PR #322 (2022-10-20): clarify the iteration order of IndexMap / IndexSet
closes #321
closed 2022-10-20 by bors[bot]