rust-embedded/heapless #330: Is there interest for `heapless::CString`?

vrmiguel, 2022-12-26:
I believe there's an interesting use-case for `heapless::Vec` to represent always null-terminated strings for FFI purposes

I've done something like this before in [unixstring](https://github.com/vrmiguel/unixstring) so I believe I could make a PR for this if the maintainers would like this functionality

```rust
// How it could look

let mut path = heapless::CString<128>;

path.append("/home/")?;
path.append(get_username())?;
path.append("/.config/my_app")?;

let file_exists = nix::unistd::access(path.as_cstr(), AccessFlags::F_OK)?;

jeandudey, 2023-01-20:
I'd be in favor of something like this, FFI helper types are pretty much lacking in bare-metal environments

t-moe, 2023-09-26:
I would also be interested in this...

PR #549 (2025-04-15): Add `CString` cont.
This PR is a continued work of #342, the heapless implementation of [`alloc::ffi::CString`](https://doc.rust-lang.org/stable/alloc/ffi/struct.CString.html). Credits to @vrmiguel.

Closes #330.

# Notable changes from the previous PR

- Implement `Deref<Target = CStr>` for `CString`.
  - As well as `Default`, `AsRef`, `Borrow`, `PartialEq`, `Eq`, `PartialOrd`, and `Ord`
- Remove the `len` and `as_str_unchecked` associated functions that should be implemented in the upstream of `core::ffi::CStr`.
- Rename the `push_bytes` associated function `extend_from_bytes`.
  - This chan [... cut]
closed 2025-04-15 by reitermarkus