rust-embedded/heapless #356: Is it the rust equivalent of the "Embedded Template Library" in C++?

jerabaul29, 2023-03-11:
I see the "Embedded Template Library" (ETL, https://github.com/ETLCPP/etl ) being used quite often for example in embedded programming in C++.

A question: is this crate well the "Rust equivalent" of the ETL? If so, could this be expressed in a short sentence in the documentation so people used for a long time to the ETL understand at once what this crate is? :)

burrbull, 2023-03-11:
Why this should be "equivalent" of something?

jerabaul29, 2023-03-11:
Maybe the wording is not the best, but the question is - does this crate and ETL well provide the same kind of functionalities?

t-moe, 2023-10-31:
Heapless and etlcpp both provide statically-allocated String- and other Container-Types. Etlcpp goes a bit further and provides all sorts of other classes, to act as stdlib replacement. In essence, while both libraries aim to provide data structures for systems where dynamic memory allocation is undesirable or unavailable, etlcpp is a C++ library with a broader feature set, while heapless is a Rust-specific solution focusing on a subset of commonly used data structures.