rust-embedded/heapless #359: String with truncating behavior

lelongg, 2023-03-13:
It might be desirable to have a string with truncation behavior when it is written beyond its capacity.

reitermarkus, 2024-01-21:
Can you clarify what the use case would look like for this?

lelongg, 2024-02-05:
A use case would be to have a non-panicking behavior when the string content is not vital, for logs for example. One might prefer to not have the full content and still carry on.

newAM, 2024-02-06:
I agree not panicking is useful.  I think we nay have fixed this since the issue was opened. Is this covered by the fix for #381?

I don't think we have any append  or conversion methods in `String` that panic anymore, but I am on my phone so I haven't checked.

lelongg, 2024-02-07:
The particular use case I had was with the `Write` trait which can only panic on overflow.

reitermarkus, 2025-04-03:
Closing, since we won't change the `Write` implementation to behave this way. I think the behaviour can be implemented externally using `StringInner::as_mut_vec`, `StringInner::push_str` or `StringInner::push`.

closed 2025-04-03 by reitermarkus