rust-embedded/heapless #379: Deserializing a string that exceeds the length: truncate?

gauteh, 2023-09-22:
Hi,

I would like to truncate or set as None, or somehow accept without Error a string that is longer than the capacity of the String. Or is there some other way to deal with this in serde that you know of?

Related to: https://github.com/japaric/heapless/issues/359

Regards, Gaute

Frostie314159, 2023-09-25:
Bump. Any implementation of the `FromIterator` trait should never panic. We should probably settle on a better behavior in the future, like truncation.

newAM, 2023-10-30:
> Any implementation of the `FromIterator` trait should never panic.

I agree that this is not desirable, but I am curious, is this a requirement?  I took a look at the documentation but did not see anything.

Dirbaio, 2023-10-30:
These are two independent issues:

- FromIterator from a too long iterator panics.
- Serde deserialization return an error, doesn't panic.

This issue is about the 2nd, I guess?

IMO the default for deserializing a long string should be an error. Other behaviors (like truncating) should be opt-in. So I wouldn't change the current serde impls. 

Truncating can be done with a wrapper, which doesn't necessarily have to be in the heapless crate. For example you can write your own parser and use the serde [`with` attribute](https://serde.rs/field-attrs.html#with).