rust-embedded/heapless #380: Alignment of internal arrays

MabezDev, 2023-09-29:
I often reach for `heapless::Vec`, but find myself having to write a custom slice wrapper due to peripheral alignment constraints have you had any thoughts about introducing a way to constrain the internal buffers alignment some how?

I found a quite nice crate called [elain](https://docs.rs/elain/0.3.0/elain/) which I used in [embedded-fatfs' block device wrapper](https://github.com/MabezDev/embedded-fatfs/blob/b05b9301679e66bf203796f24a5b4d4e546ee239/src/device.rs#L154-L161). It allows the alignment of a field to be defined by a const generic parameter.

JohnScience, 2023-12-24:
I would love this feature too.

JohnScience, 2023-12-26:
@MabezDev https://github.com/rust-embedded/heapless/pull/429