rust-embedded/heapless #382: Maintenance status?

Dirbaio, 2023-10-10:
PR merging and releases have slowed down a bit recently. nothing merged since January this year.

I'd be great to get maintenance unstuck, heapless is a great crate and very widely used. I volunteer myself, hopefully other people will too.

To increase the bus factor, I propose moving the repo to [rust-embedded-community](https://github.com/rust-embedded-community/). Adding individual maintainers helps, but they can always get bus'd, which is what rust-embedded-community exists to avoid.

What do you think @japaric ?

gauteh, 2023-10-12:
This is a very useful crate. Thanks for making it! I hope you find some solution to help maintaining it, you probably have a significant amount of users by now.

newAM, 2023-10-20:
Fixed in https://github.com/rust-embedded/wg/issues/710 🎉

closed 2023-10-20 by newAM