rust-embedded/heapless #428: [Feature]: String to Integer functionality

Murmele, 2023-12-21:
It would be nice to have a possibility to convert a string to an integer and vice versa. Do you have already a plan to implement something like this? Otherwise if you want, I can implement it?

newAM, 2023-12-21:
Can you expand on what you mean by this, perhaps provide an example function signature and explain what encodings would be used?

The `String` type in `heapless` is UTF-8 encoded which can have a variable number of bytes per character.  For conversion to bytes we have the `into_bytes` method.

Dirbaio, 2023-12-21:
you can convert from heapless::String to integer with `u32::from_str(&my_heapless_string)`. see [docs](https://doc.rust-lang.org/std/str/trait.FromStr.html#tymethod.from_str).

You can convert from integer to heapless::String with `heapless::format!("{}", my_int)`.

is this what you were looking for?

Murmele, 2023-12-24:
Thanks this looks good.

closed 2023-12-24 by Murmele