rust-embedded/heapless #431: Add DoubleEndedIterator to HistoryBuffer::oldest_ordered

Eugeny, 2023-12-28:
Would be very useful and seems relatively straightforward to implement ✌️

Eugeny, 2023-12-29:
Dumb implementation:

```rust

pub struct HistoryBufferDoubleEndedIterator<'a, T, const N: usize> {
    buf: &'a HistoryBuffer<T, N>,
    cur: usize,
}

impl<'a, T, const N: usize> HistoryBufferDoubleEndedIterator<'a, T, N> {
    pub fn new(buf: &'a HistoryBuffer<T, N>) -> Self {
        Self { buf, cur: 0 }
    }
}

impl<'a, T, const N: usize> ExactSizeIterator for HistoryBufferDoubleEndedIterator<'a, T, N> {}

impl<'a, T, const N: usize> Iterator for HistoryBufferDoubleEndedIterator<'a, T, N> {
    type Item = &'a T;

    fn next(&mut self) -> Option<&'a T> {
        let (a, b) = self.buf.as_slices();
        self.cur += 1;
        if self.cur < a.len() {
            Some(&a[self.cur])
        } else if self.cur < a.len() + b.len() {
            Some(&b[self.cur - a.len()])
        } else {
            None
        }
    }

    fn size_hint(&self) -> (usize, Option<usize>) {
        (self.buf.len(), Some(self.buf.len()))
    }
}

impl<'a, T, const N: usize> DoubleEndedIterator for HistoryBufferDoubleEndedIterator<'_, T, N> {
    fn next_back(&mut self) -> Option<Self::Item> {
        let (a, b) = self.buf.as_slices();
        if self.cur == 0 {
            return None;
        }
        self.cur -= 1;
        if self.cur < a.len() {
            Some(&a[a.len() - self.cur - 1])
        } else if self.cur < a.len() + b.len() {
            Some(&b[b.len() - self.cur - 1])
        } else {
            None
        }
    }
}

```

witek103, 2024-01-25:
you can look at https://github.com/rust-embedded/heapless/pull/449

Dirbaio, 2024-07-01:
done in #449

closed 2024-07-01 by Dirbaio