rust-embedded/heapless #450: Heapless String doesn't implement From trait

lvanasse, 2024-01-26:
Hi, I'm fairly new to Rust, so I apologize in advance if my issue is not clear or not exact.

I've being trying to use `serde-json-core` with Zenoh and I get this error when I try to publish a simple struct with a String and an u64.

```
error[E0277]: the trait bound `Value: From<serde_json_core::heapless::String<_>>` is not satisfied
   --> src/nodes/node_example2.rs:39:27
    |
39  |             publisher.put(serde_json_core::to_string(&node_msg_example.into()).unwrap()).res_sync().unwrap();
    |                       --- ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `From<serde_json_core::heapless::String<_>>` is not implemented for `Value`
    |                       |
    |                       required by a bound introduced by this call
    |
    = help: the following other types implement trait `From<T>`:
              <Value as From<isize>>
              <Value as From<i8>>
              <Value as From<i16>>
              <Value as From<i32>>
              <Value as From<i64>>
              <Value as From<usize>>
              <Value as From<u8>>
              <Value as From<u16>>
            and 13 others
    = note: required for `serde_json_core::heapless::String<_>` to implement `Into<Value>`
note: required by a bound in `Publisher::<'a>::put`
   --> /home/ludovic/.cargo/registry/src/index.crates.io-6f17d22bba15001f/zenoh-0.10.1-rc/src/publication.rs:415:20
    |
413 |     pub fn put<IntoValue>(&self, value: IntoValue) -> Publication
    |            --- required by a bound in this associated function
414 |     where
415 |         IntoValue: Into<Value>,
    |                    ^^^^^^^^^^^ required by this bound in `Publisher::<'a>::put`

For more information about this error, try `rustc --explain E0277`.
```
From my understanding, the String implementation in heapless needs to satisfied the From trait. Although maybe I didn't create my  
string properly or didn't give it properly to Zenoh function. If you have any insight into that, it would be much appreciated.

Here's what I'm trying to publish on Zenoh:

```rust
use serde::{Deserialize, Serialize};

#[derive(Serialize, Deserialize)]
pub struct NodeMsgExample<'a> {
    pub msg_str: &'a str,
    pub msg_random_num: u64,
}

```

And here's the code I'm using to publish the NodeMsgExample:
```rust
let mut config = Config::default();
config.transport.shared_memory.set_enabled(true).unwrap();
println!("Opening session...");
let session = zenoh::open(config).res_sync().unwrap();

let publisher = session
    .declare_publisher("test/node")
    .congestion_control(CongestionControl::Block)
    .res_sync()
    .unwrap();

let node_msg_example = NodeMsgExample {
    msg_str: "node_2",
    msg_random_num: self.n,
};
loop {
    println!(
        "Publish on node 2: {str} {num}",
        str = node_msg_example.msg_str,
        num = node_msg_example.msg_random_num
    );

    // publisher.put(serde_json_core::to_string(&node_msg_example).unwrap()).res_sync().unwrap();
    publisher.put(serde_json_core::to_string(&node_msg_example).unwrap()).res_sync().unwrap();
    thread::sleep(Duration::from_secs(1));
}
```

If you need more information from me, don't hesitate to ask.

Thanks in advance, cheers!

newAM, 2024-01-27:
I was not able to reproduce this.  I put my code here: https://github.com/newAM/heapless-issue-450

I did use the latest (unreleased) git version of `serde_json_core` because the current release is using an older heapless (`0.7`), which might be why I couldn't reproduce it.

Can you share your `heapless`/`serde`/`serde_json_core` versions?

lvanasse, 2024-01-29:
Hi @newAM, thanks a lot for your quick answer. 

I've reused the same version as you've proposed in your repository. Which are:
```toml
heapless = { version = "0.8.0", features = ["serde"] }
serde = { version = "1.0.196", features = ["derive"] }
serde-json-core = { git = "https://github.com/rust-embedded-community/serde-json-core", rev = "9327a14e74ad3b4fd37a9ac34e72b61aa5fcc9bf" }
```

But I still get this error:

```bash
error[E0277]: the trait bound `Value: From<serde_json_core::heapless::String<_>>` is not satisfied
   --> src/nodes/node_example2.rs:38:27
    |
38  |             publisher.put(serde_json_core::to_string(&node_msg_example).unwrap()).res_sync().unwrap();
    |                       --- ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `From<serde_json_core::heapless::String<_>>` is not implemented for `Value`
    |                       |
    |                       required by a bound introduced by this call
    |
    = help: the following other types implement trait `From<T>`:
              <Value as From<isize>>
              <Value as From<i8>>
              <Value as From<i16>>
              <Value as From<i32>>
              <Value as From<i64>>
              <Value as From<usize>>
              <Value as From<u8>>
              <Value as From<u16>>
            and 13 others
    = note: required for `serde_json_core::heapless::String<_>` to implement `Into<Value>`
note: required by a bound in `Publisher::<'a>::put`
   --> /home/ludovic/.cargo/registry/src/index.crates.io-6f17d22bba15001f/zenoh-0.10.1-rc/src/publication.rs:415:20
    |
413 |     pub fn put<IntoValue>(&self, value: IntoValue) -> Publication
    |            --- required by a bound in this associated function
414 |     where
415 |         IntoValue: Into<Value>,
    |                    ^^^^^^^^^^^ required by this bound in `Publisher::<'a>::put`

For more information about this error, try `rustc --explain E0277`.
```

Maybe I'm not understanding the error correctly? Maybe the issue might be with serde-json-core? 

Sorry, I'm a bit confused :confused:

newAM, 2024-01-29:
Ah I see the issue now, zenoh doesn't implement conversion from heapless types for Value: https://docs.rs/zenoh/0.10.1-rc/zenoh/value/struct.Value.html

Try converting the heapless String to a str first.

I'm on my phone now so I can't easily provide an example.  If this doesn't help let me know and I'll provide a code example when I have a proper keyboard.

lvanasse, 2024-01-29:
@newAM sadly I still get the same error.

Here's what I'm using now:

```rust
use serde::{Deserialize, Serialize};

#[derive(Serialize, Deserialize)]
pub struct NodeMsgExample<'a> {
    pub msg_str: &'a str,
    pub msg_random_num: u64,
}
```

From my understanding, this would be a statically allocated string when I create my NodeMsgExample struct (?). Which, I don't think I'm allocating anything static here? Although there might be some Rust black magic going on.

```rust
let mut config = Config::default();
config.transport.shared_memory.set_enabled(true).unwrap();
println!("Opening session...");
let session = zenoh::open(config).res_sync().unwrap();

let publisher = session
    .declare_publisher("test/node")
    .congestion_control(CongestionControl::Block)
    .res_sync()
    .unwrap();

let node_msg_example = NodeMsgExample {
    msg_str: "node_2",
    msg_random_num: self.n,
};
loop {
    println!(
        "Publish on node 2: {str} {num}",
        str = node_msg_example.msg_str,
        num = node_msg_example.msg_random_num
    );

    publisher.put(serde_json_core::to_string(&node_msg_example).unwrap()).res_sync().unwrap();
    thread::sleep(Duration::from_secs(1));
}
```

Again, sorry for my lack of expertise in Rust, I'm learning as I go :sweat_smile:

newAM, 2024-01-30:
No worries :)

Currently the type going into `publisher.put` is `heapless::String`, but `zenoh::value::Value` doesn't implement `From<heapless::String>`, instead try `serde_json_core::to_string(&node_msg_example).unwrap().as_str()`, the final `.as_str()` changes the representation to a `&str`.

newAM, 2024-02-04:
That's correct, the heapless types need a size for the static buffer they use, it's a little bit more clear what's going on when expanding the code:

```rs
#[derive(serde::Serialize, serde::Deserialize)]
pub struct NodeMsgExample<'a> {
    pub msg_str: &'a str,
    pub msg_random_num: u64,
}

fn main() {
    let node_msg_example = NodeMsgExample {
        msg_str: "node_2",
        msg_random_num: 123456,
    };

    let json_heapless: heapless::String<32> =
        serde_json_core::to_string(&node_msg_example).unwrap();

    let json_str: &str = json_heapless.as_str();

    let value = zenoh::value::Value::from(json_str);
}
```

lvanasse, 2024-04-01:
Sorry for the delay @newAM, this works for me! Thank you very much for all your help, it is very much appreciated!

closed 2024-04-01 by lvanasse