rust-embedded/heapless #454: `object_pool!()` macro requires both a CamelCase and SNAKE_CASE name

cbrune, 2024-02-13:
If you want to give your pool a descriptive name, like `MyMemPool`, the `object_pool!()` macro will not allow it.

The examples in the documentation side step this issue by using single character identifiers for the pool.

For example using `MyMemPool`:
```rust
object_pool!(MyMemPool: [u8; 128]);
```

Gives this failure:
```
error: static variable `MyMemPool` should have an upper case name
  --> test.rs:41:18
   |
41 |     object_pool!(MyMemPool: [u8; 128]);
   |                  ^^^^^^^^^ help: convert the identifier to upper case: `MY_MEM_POOL`
   |
   = note: `-D non-upper-case-globals` implied by `-D bad-style`

error: aborting due to previous error
```
Then if I use a SCREAMING_SNAKE_CASE identifier,
```rust
object_pool!(MY_MEM_POOL: [u8; 128]);
```
 I get this failure:
```
error: type `MY_MEM_POOL` should have an upper camel case name
  --> test.rs:41:18
   |
41 |     object_pool!(MY_MEM_POOL: [u8; 128]);
   |                  ^^^^^^^^^^^ help: convert the identifier to upper camel case: `MyMemPool`
   |
   = note: `-D non-camel-case-types` implied by `-D bad-style`

error: aborting due to previous error
```

Looking at the macro impl, it looks like the thing that user will interact with is the `struct $name`, so the identifier passed into `object_pool` should be a CamelCase identifier.  Inside the macro when creating the static variable, convert `$name` to SCREAMING_SNAKE_CASE.

The `paste` crate is useful for situations like this: https://docs.rs/paste/1.0.14/paste/

Dirbaio, 2024-02-13:
We should avoid `paste` because proc macros. 

IMO a simpler fix is to choose either CamelCsae or YELLING_CASE and add an `#[allow(..)]` to the other to silence the warning.

reitermarkus, 2024-02-19:
This is already fixed by https://github.com/rust-embedded/heapless/pull/420 but not released yet. Also, see https://github.com/rust-embedded/heapless/pull/417#issuecomment-1823044567.

closed 2024-02-19 by reitermarkus