rust-embedded/heapless #507: [construct|deconstruct] Vec/String [from|to] length + [MaybeUninit<T>; N]

Felix-El, 2024-08-01:
These operations, let's call them `from_parts` / `into_parts`, I believe would be quite useful to allow the payload to change owners/container types without moving the data.

I don't think `into_array<M>()` can achieve the same because one has to statically know the exact runtime size.
I thought about permitting `M` to be up to the capacity `N` but into_array would then have to pad the unused remaining space, which may be quite inefficient.

What are the chances of such MR to be accepted?

Dirbaio, 2024-08-01:
> I believe would be quite useful to allow the payload to change owners/container types without moving the data.

functions taking or returning the full `[MaybeUninit<T>; N]` will still move the data, I'm not sure if adding these functions will allow any performance improvements.

reitermarkus, 2025-04-27:
Closing, since this would still move the data.

Feel free to re-open with a concrete use case example.

closed 2025-04-27 by reitermarkus